cta-observatory/ctapipe #1088: Future updates in scikit-learn could brake behaviour of energy reconstruction

HealthyPear, 2019-06-17:
In the framework of use of _protopipe_ (or any other pipeline which makes use of basic _ctapipe_'s functionalities to estimate the energy of showers), a call to _ctapipe_'s `energy_regressor.py` is made.
During such operation, the 'joblib' method is imported explicitly from scikit-learn (line #300 in `reco/regresson_classifier_base.py` - at least).

I recently got a warning explaining that this method will be removed in future versions of the library- possibly because it will become (or is already) a separate package.

With this issue I then wish to report that if _ctapipe_'s environment gets updated, it might stop working.

Running from the _GRID_, I got many empty files because of this: apparently there the _sciki-learn_ package has been already updated.
I then suggest to update the outdated line (and any other one which does the same) to `import joblib` (of course, you would also need to add the _joblib_ package to the list of required packages in `setup.py`).

MaxNoe, 2019-06-17:
@HealthyPear could you write a minimal example that raises this warning?

HealthyPear, 2019-06-17:
Dear Maximilian,

I am afraid "minimal" is a big approximation :(
For the moment I try to give you at least the Traceback of the issue that I got from DIRAC.

During the generation of DL1 data for classification in protopipe, the estimated energy is used.
When doing this, from `write_dl1.py` the following piece of code is selected,

`from ctapipe.reco.energy_regressor import *` (this is bad, I know...)

`regressor = EnergyRegressor.load(reg_file, cam_id_list=args.cam_ids)`

where `args.cam_ids` is something like "LSTcam" and `reg_file` is the file generated by the script `build_model.py` (please, refer to the - now public - relevant protopipe's documentation / code).

From here the rest of the Traceback is fully-ctapipe dependent (mind that here the relevant ctapipe is the one on the DIRAC environment, which has been evidently updated),

```
  File "/cvmfs/cta.in2p3.fr/software/miniconda/envs/ctapipe_v0.6.2/lib/python3.6/site-packages/ctapipe/reco/energy_regressor.py", line 156, in load
    return super().load(path, cam_id_list, unit)
  File "/cvmfs/cta.in2p3.fr/software/miniconda/envs/ctapipe_v0.6.2/lib/python3.6/site-packages/ctapipe/reco/regressor_classifier_base.py", line 311, in load
    self.model_dict[key] = joblib.load(path.format(cam_id=key))
  File "/cvmfs/cta.in2p3.fr/software/miniconda/envs/ctapipe_v0.6.2/lib/python3.6/site-packages/sklearn/externals/joblib/numpy_pickle.py", line 598, in load
    obj = _unpickle(fobj, filename, mmap_mode)
  File "/cvmfs/cta.in2p3.fr/software/miniconda/envs/ctapipe_v0.6.2/lib/python3.6/site-packages/sklearn/externals/joblib/numpy_pickle.py", line 526, in _unpickle
    obj = unpickler.load()
  File "/cvmfs/cta.in2p3.fr/software/miniconda/envs/ctapipe_v0.6.2/lib/python3.6/pickle.py", line 1050, in load
    dispatch[key[0]](self)
  File "/cvmfs/cta.in2p3.fr/software/miniconda/envs/ctapipe_v0.6.2/lib/python3.6/pickle.py", line 1338, in load_global
    klass = self.find_class(module, name)
  File "/cvmfs/cta.in2p3.fr/software/miniconda/envs/ctapipe_v0.6.2/lib/python3.6/pickle.py", line 1388, in find_class
    __import__(module, level=0)
ModuleNotFoundError: No module named 'joblib'
```

vuillaut, 2019-06-17:
Well no big deal, just `import joblib` instead of `from sklearn.externals import joblib`

MaxNoe, 2019-06-17:
It's kind of a big deal because scikit learn came with it's own version of joblib so a fixed version of joblib was guaranteed for a fixed version of scikit learn. If this is no longer the case, we need to fix a version of joblib to ctapipe or a processing so we can load the models back.

MaxNoe, 2019-06-17:
Dear Maximilian,

> I am afraid "minimal" is a big approximation :(
For the moment I try to give you at least the Traceback of the issue that I got from DIRAC.

I was asking about the changes in sklearn.

So am I right that the minimal example would be

```python

from sklearn.externals import joblib
```

?

vuillaut, 2019-06-17:
I see.

This is weird though because I can still `from sklearn.externals import joblib` with the latest (0.20.3) version of scikit-learn... it's more an installation issue I think?

vuillaut, 2019-06-17:
@MaxNoe 
Concerning the version issue, if `joblib` is a dependency of `sklearn`, the right version will be installed when scikit-learn is installed, no?
And we don't need to add joblib to ctapipe's dependency list.

vuillaut, 2019-06-17:
For ref:
https://github.com/scikit-learn/scikit-learn/pull/13531

MaxNoe, 2019-06-17:
> Concerning the version issue, if joblib is a dependency of sklearn, the right version will be installed when scikit-learn is installed, no?

The *problem* ja, sklearn is basically compatible  with any joblib version but a model has to be loaded with the same joblib version that was used to dump it.

So the dependency is not fixed for sklearn, but it might make sense to fix it for ctapipe to make sure a model that was trained with a certain version of ctapipe is loadable with that ctapipe version. 

At least for a specific analysis the joblib version must be fixed.

MaxNoe, 2019-06-17:
Also from the joblib docs:

> As of Python 3.8 and numpy 1.16, pickle protocol 5 introduced in PEP 574 supports efficient serialization and de-serialization for large data buffers natively using the standard library:

> pickle.dump(large_object, fileobj, p

kosack, 2019-06-18:
This implies another issue as well: we should use a standard (and non-version-specific) way of storing models.  I recall that @mackaiver had used a library that writes them as some industry-standard XML or something.  

In any case, we can try to specifically fix the version of joblib and sklearn in both `setup.py` and in the conda package to avoid this problem.

MaxNoe, 2019-06-18:
> This implies another issue as well: we should use a standard (and non-version-specific) way of storing models. I recall that @mackaiver had used a library that writes them as some industry-standard XML or something.

Yes, we used pmml for storing models to be interoperable with java:
https://de.wikipedia.org/wiki/Predictive_Model_Markup_Language

This is however considered obsolete by the people who developed it but for the successor, there is basically no documentation and no scikit-learn plugin.

Both projects seem rather dead at this point, which is a shame.

bregeon, 2019-06-19:
As documentation, here are a few links (from discussion with Max):

- https://www.andrey-melentyev.com/model-interoperability.html#designated-model-interchange-formats
 - https://github.com/onnx/onnx
 - https://en.wikipedia.org/wiki/Predictive_Model_Markup_Language
 - http://dmg.org/pmml/v4-3/GeneralStructure.html
 - http://github.com/fact-project/aict-tools

kosack, 2019-06-19:
It's too bad PMML/PFA seem to be dead - those looked nice.  I guess ONNX is the way to go, though it seems more complex as it's not just a format, but also a runtime for predicting... It would be nicest to just store in ONNX format and then read back and use sklearn (leaving open the option to also use their runtime in the future)

vuillaut, 2019-06-19:
Without any easy dump format/solution, wouldn't it be easier to just fix a joblib version in ctapipe?

kosack, 2019-06-19:
I just checked and I don't see why we are having this problem. The latest scikit-learn package automatically depends on joblib, so it's hard to have it installed without joblib there already:

```
conda update scikit-learn                 (cta-dev)  
Collecting package metadata: done
Solving environment: done

## Package Plan ##

  environment location: /Users/kkosack/anaconda3/envs/cta-dev

  added / updated specs:
    - scikit-learn
...
The following NEW packages will be INSTALLED:

  joblib             conda-forge/noarch::joblib-0.13.2-py_0
 ```

What I will do right now is create a new package for ctapipe-0.6.2 that fixes the version of scikit-learn to 0.21.3 (which seems to be the latest), and thus also installs a compatible joblib.  Once that's done, on the grid we can update the environment and it should all work.

kosack, 2019-06-19:
Or we can stick with 0.20.3, which doesn't need an external joblib package (either way is fine as long as we are consistent)

vruusmann, 2019-06-19:
> It's too bad PMML/PFA seem to be dead - those looked nice

The rumors about PMML's death are greatly exaggerated. There's a minor PMML 4.4 schema version update coming up shortly, with a major PMML 5.0 schema version upgrade already in the works (the development happens on a legacy SourceForge and/or http://mantis.dmg.org environments, not GitHub).

The liveliness of a standard does not depend on its regulatory body (DMG.org is a closed group, and based on my recent experience, they are not accepting new members), but on the people using it to solve their everyday problems. The PMML standard is doing very nicely in this regard.

MaxNoe, 2019-06-19:
> I just checked and I don't see why we are having this problem. 

*We* are not having this problem I suppose. 

The way to go forward is:
Require joblib or a (as soon as released) a scikit-learn version that depends explicitly on joblib

 `from joblib import load, dump`

instead of `from sklearn.externals import joblib`

Problems can only arise when doing the first with an old scikitlearn version or the second with a newer scikit learn version. The first case would be remedied by explicitly requiring joblib.
(which we should do, we import it directly)

bregeon, 2019-06-19:
@kosack , indeed, there is no real problem from the point of view of joblib version, or joblib being in or outside sklearn. The question raise by Michele came from the fact that he is building his models with ctapipe v0.6.1 (because he cannot install v0.6.2 on his Mac because of the missing eventio package) and then needs to run his models on the grid with v0.6.2 that has a different version of joblib, hence the crash.
As you have opened #1091 , and as there is no real problem, I think this issue can be closed and probably we can try to fix Michele's specific problem on the redmine issue: https://forge.in2p3.fr/issues/37680

HealthyPear, 2019-06-20:
@bregeon Yesterday Karl uploaded the version 0.11 of _eventio_ on _Anaconda Cloud_, hence I was able to install the stable version of _ctapipe_ 0.6.2 from there (now I should have the same version as on DIRAC).

A problem in any case remains, as far as I understand: either we freeze the version of _scikit-learn_ or we have to have find a way to keep up with the changes.
This is because 1) of the _joblib_ dependence, since newer versions of _sklearn_ drop it 2) because even if maybe now the difference is small betwee _sklearn_ 0.20 and 0.21, perhpas in the future the difference will be bigger and models pickled from an older version could stop being imported or (even worst) give wrong results.

MaxNoe, 2020-03-20:
This is fixed, master used joblib directly

closed 2020-03-20 by MaxNoe