cta-observatory/ctapipe #1090: Calculate Hillas Parameters in TelescopeFrame

MaxNoe, 2019-06-18:
This was a comment made by Razmik at the meeting in Lugano and something I heard from many people.

We should calculate hillas parameters in telescope frame, so that the units are deg and comparable between telescopes.

thomasgas, 2019-06-18:
Easy fix: divide the values that we have in meters by the focal length, which gives the result in radians that can be converted to deg?

kosack, 2019-06-18:
that was the original idea in fact, just somehow never happened since the coordinate frames weren't finished until after the Reconstructor code was written (it originally did the transform internally).  We could transform the cameras first, or transform the hillas ellipse after.  Either way lets you compare.  I don't see this as a major problem right now, since you can still do that, but i guess to be able to compare histograms of width and length, it would be nice to store them that way in the DL1 output. 

If you really want to compare, it should be in something like the `NominalFrame`, since with the `TelescopeFrame` you still have a projection effect if the pointing isn't parallel, but since the HillasReconstructor does things in 3D, that frame is not needed.

kosack, 2019-06-18:
> Easy fix: divide the values that we have in meters by the focal length, which gives the result in radians that can be converted to deg?

Isn't that basically the transform from Camera to Telescope? You could also load all of the hillas.x and hillas.y values (already computed)  into a SkyCoord and transform to the TelescopeSystem for each telescope and have the same effect.  The question is what is the most useful for storage in DL1? If you later re-measure the real focal length and find it is different from the one you used, you'd have to redo the transform to camera and back again to fix it, but I guess you'd also start from DL0 data in that case, so it wouldn't matter.

kosack, 2019-06-18:
Of course dealing with width and length, etc,  is slightly different (we'd have to have a transform function for lengths, not just points), so maybe just using the Telescope or Nominal frame to start with is easier to understand (and then the conversion can be removed from the HillasReconstructor).

MaxNoe, 2019-06-18:
> Of course dealing with width and length,

Not really. As we are dealing with small angles (< 10 degrees for all telescopes).

kosack, 2019-06-18:
Well, you still need a simple function to convert from meters to degrees. Not difficult, I agree, but right now you can only transform points, not a HillasParametersContainer. Wouldn't be hard to do though.

kosack, 2019-06-18:
I guess this issue boils down to: *can you make a histogram of Hillas width for all telescopes in such a way as you can compare them, from DL1 output alone?*,

 and I think the answer is *yes, with a coordinate transform*. Right?  So either we close it, or suggest that it would be easier to do if we stored them in a different Frame.

maxnoe, 2021-12-06:
Done

closed 2021-12-06 by maxnoe