cta-observatory/ctapipe #1097: How redirect to a file the logging messages of a tool?

FrancaCassol, 2019-06-25:
Hi,

I would like to redirect to a log file the logging messages of a ctapipe tool (avoiding the unix redirection), which is the best way of doing it? 

Thanks

MaxNoe, 2019-06-25:
add something like 

```
import logging

log = logging.getLogger('ctapipe')
logfile = logging.FileHandler('/path/to/togfile')
log.addHandler(logfile)
```

to your script.

kosack, 2019-06-25:
probably we should put an option to do that in the `Tool` base class as well, so that all tools can do it.

kosack, 2019-06-25:
I haven't tried it but you might have to remove the default StreamHandler if you don't want stdout *as well as the file* in @MaxNoe's example above (assuming addHandler() doesn't replace the old one by default).  Otherwise, you get it both to stdout and to the file you specified.

kosack, 2019-06-26:
If Max's suggestion works, can we close this?

FrancaCassol, 2019-06-26:
Hi @kosack,

Yes, it works but I do not menage to eliminate the stdout. I did this but it seems not to work, do you have other suggestions?

```
            log.removeHandler(logging.StreamHandler(sys.stdout))
            log.removeHandler(logging.StreamHandler(sys.stderr))

```
Also the output to file has some problem with the fonts:

[1;32mINFO[0m [CalibrationHDF5Writer] (tool/initialize): ctapipe version 0.6.2.post161+gitc154718
[1;32mINFO[0m [CalibrationHDF5Writer] (tool/run): Starting: CalibrationHDF5Writer
[1;32mINFO[0m [CalibrationHDF5Writer.FlasherFlatFieldCalculator] (flatfield/__init__): extractor <ctapipe.image.extractor.LocalPeakWindowSum object at 0x7f4dbc180f60>

MaxNoe, 2019-06-26:
This is not a Problem with fonts, this includes color escape sequences.
We should set a formatter that does not include them.

For the moment you can look into the files with `less -R` which will show the colors correctly.

The code you showed cannot work, because you are trying to remove a freshly created handler.
I managed to silence the logging output do console by using:

```
h = logging.StreamHandler()
logging.addHandler(h)
h.close()
```

FrancaCassol, 2019-06-26:
Clever idea, but it doesn't work for me :-( 
Inside my tool I do:
```
    def setup(self):
        kwargs = dict(parent=self)

        if self.log_file!='None':
            log = logging.getLogger('CalibrationHDF5Writer')
            logfile = logging.FileHandler(self.log_file)
            log.addHandler(logfile)

            h = logging.StreamHandler()
            h.setLevel(logging.FATAL + 1)
            log.addHandler(h)

```
perhpas there are other streamhandlers defined ? how can one get the list of the active handlers?

MaxNoe, 2019-06-26:
>  if self.log_file!='None':

That probably should be `self.log_file is not None`

FrancaCassol, 2019-06-26:
it could be but this is not my problem, I get the log file, but also the stdout

kosack, 2019-06-27:
I think you have to delete the default StreamHandler first. What you have will send the logs to both places. You might be able to something like `logger.removeHandler(logger.handlers.pop())`

kosack, 2019-06-27:
for the color, you can disable it by changing the `LogFormatter` (which we set to ColorFormatter by default in the Tool). I like having the color output for the console, so you couljd just change it to a plain one in the case that a log file is used.

MaxNoe, 2019-06-27:
As the formatter is attached to the handler, this is no problem.

Just use the `ColorFormatter` for `StreamHandler` and the plain formatter for the `FileHandler`

MaxNoe, 2019-06-27:
We could add a `-q, --quiet` option to silence the StreamHandler and a `--logfile` option for writing a logfile to every tool.

maxnoe, 2021-02-19:
Implemented together with the possibility to extensively configure logging through the tool config in #1310

closed 2021-02-19 by maxnoe