cta-observatory/ctapipe #1100: Getting uncertainties on time gradient and intercept in timing_parameter

andres-baquero, 2019-07-09:
Hi, I'm triying to get the uncertainties of time gradient and intercept for using in RF for reconstruction.

I understand that in numpy.polynomial.polynomial.polyfit function  the parameter singular_values returns the singular values of the scaled Vandermonde matrix, with contains the uncertainties, in this case for intercept and slope (time gradient). But now I'm getting just a matrix of [1. 1.].  And this matrix Vandermonde must be of 2x2.

Which is the best way of doing it?

Thank you.

MaxNoe, 2019-07-09:
You could use `numpy.polynomial.Polynomial.fit`.

polyfit is deprecated anyways.

MaxNoe, 2019-07-09:
You could use https://en.wikipedia.org/wiki/Weighted_least_squares

Also: maybe the uncertainties on the parameters are interesting, but also the sum / max / min of the residuals could be very helpful

MaxNoe, 2019-07-10:
Fixed by merging #1101

closed 2019-07-10 by MaxNoe