cta-observatory/ctapipe #1102: import ctapipe modules time

vuillaut, 2019-07-09:
Some modules take a long time to import.
e.g.:
```
In [1]: %time import ctapipe.image
CPU times: user 22.2 s, sys: 471 ms, total: 22.6 s
Wall time: 22.4 s
```

Is this normal? It seems unexpectedly long.
I checked on two different machines and get similar results.

I have trouble tracking down the reason of it, any help or explanation would be welcome.

MaxNoe, 2019-07-09:
This seems to be compilation in `ctapipe/image/extractor.py`.

If I remove `from .extractor import *` from `ctapipe/image/__init__.py` the import is done in 9.6 seconds.

vuillaut, 2019-07-09:
Couldn't the compilation be done during installation rather than import?

MaxNoe, 2019-07-09:
This is numba code, I don't think so

vuillaut, 2019-07-09:
Ha yes right.

MaxNoe, 2019-07-09:
The `neighbor_average_waveform_jit is the offender`

It compiles on import, I though numba would compile on first use???

MaxNoe, 2019-07-09:
This function is compiled for multiple input data types, If I just compile for one, the import time is reduced to 3 seconds.

For half of the signatures, there is a `TODO Deprecate`

kosack, 2019-07-10:
> It compiles on import, I though numba would compile on first use???

that used to be the behavior - maybe it changed?

kosack, 2019-07-10:
It's also strange that such a small function takes so long to compile. Perhaps it is compiling versions for many possible input types? That can be restricted, I think

MaxNoe, 2019-07-10:
It's already restricted to 8 different input combinations. 4 of them marked with `# TODO: deprecate`

kosack, 2019-07-10:
Perhaps we can restrict to one only (I think we can assume all waveforms will be the same type), so that might give some speedup.

kosack, 2019-07-10:
I think just adding `cache=True` to the decorator may solve it: 

```py3
@guvectorize(
    [
        (float64[:], int64, int64, int64, float64[:]),
        (float32[:], int64, int64, int64, float64[:]),
    ],
    '(s),(),(),()->()',
    nopython=True,
    cache=True,
)
def sum_samples_around_peak(waveforms, peak_index, width, shift, ret):

```

MaxNoe, 2019-07-10:
Could we then import it in `setup.py` so that it basically happens on install?

kosack, 2019-07-10:
Wouldn't that mean I would then need to build a package for all architectures again? I was hoping to not require that if possible

MaxNoe, 2019-07-10:
Numba compiles to llvm bytecode, that is machine independent, right?

watsonjj, 2019-07-10:
I don't think a compilation of numba methods on install is necessary.

Cache should do the trick, however in my tests with that option in the past, I did not see a reduction of import time. Have you tested it in this instance?

Of course I agree with the removal of the areas marked deprecated.

watsonjj, 2019-07-12:
> > It compiles on import, I though numba would compile on first use???
> 
> that used to be the behavior - maybe it changed?

I suppose it compiles on import if the input types are predefined. Normally it compiles on first use because it won't know the input types until called. Which has a bit of overhead because its always anticipating possible additional types. But perhaps if a module has many numba functions, it could be better to not define the input types to avoid this compilation-on-import.

kosack, 2019-07-15:
Well, if we can reduce it to a few seconds as @MaxNoe mentioned above by restricting to only one type, I think it's not a big problem.  If anyone imports the calib module, it's pretty certain that they will eventually  use that function in any case.

closed 2019-08-05 by kosack