cta-observatory/ctapipe #1109: how to do performance testing?

kosack, 2019-07-16:
The Problem
-------------

Issue #1102 (slow import of numba functions) and PR #1108 showed that the speed of importing ctapipe jumped from under a second to 10s of seconds due to a change in code, but we never noticed until much later.  That implies we need some sort of **performance testing**: measuring the speed of executing and changes between versions (see e.g. https://en.wikipedia.org/wiki/Software_performance_testing)

* Performance testing (how efficiently something runs) is not the same as unit testing (how it operates functionally). It needs history (e.g. the change since the last version), rather than a pass/fail. 
* we shouldn't include performance tests in unit tests, since at least with the current CI system, we are not guaranteed to be on the same hardware for each run. (See also https://automationpanda.com/2017/05/18/can-performance-tests-be-unit-test/) 
* How should we measure performance and automatically check for such problems?

Solutions?
----------
* This may require waiting for a CTAO-owned CI and computing platform at the SDMC (may be a while), or we could use some local resources. 
* are there cloud-based performance testing solutions we could use in the mean time?

vuillaut, 2019-07-16:
Hi Karl,

I meant to include (but haven't taken the time yet I must say) this kind of performance tests in cta-benchmark.

What do you think?

kosack, 2019-07-17:
yes, probably this is a good thing to be in the benchmarks. I didn't find any nice way to do it otherwise, though there are some cloud-based systems, they are more for testing web application performance (e.g. requests/second, etc)

kosack, 2019-07-17:
I guess in the end it implies the benchmark system will need some sort of database to keep results in, so one can compare the results between versions (both scientific and performance)