cta-observatory/ctapipe #1110: Break up `utils` into subpackages with better names

MaxNoe, 2019-07-17:
Something called `utils` will mostly end up as a catch it all module of unrelated stuff.

kosack, 2019-07-18:
current contents (which you can see are already pretty unrelated):

* ``astro``: star catalog access
* ``datasets``: access to external (supporting) data like ctapipe-extra or files in the CTAPIPE_SVC_PATH
* ``CutFlow``: class to apply and track cuts applied to events
* ``fitshistogram``: class to create and manage N-D histograms, with read/write to FITS
* ``json2fits``: convert JSON docs to FITS headers (not used?) 
* ``linalg``: linear algebra helper functions
* ``dynamic_class``:  construct a class by name (I think). Doesn't seem to be used anymore...
* ``table_interpolator, table_interpolator, template_network_interpolator, unstructured_interpolator``: general ND interpolators used by ImPACT?
* ``rgbtohex``: color conversion for e.g. web guis using Bokeh

MaxNoe, 2019-07-18:
> I think it was for serializing config files to FITS)

God help us

MaxNoe, 2019-07-18:
Had a look at linalg:

* rotation_matrix_2d: should go into the coordinate trafos
* length: not used
* normalize and angle: verbatim copies of the same functions in  `HillasReconstructor` and not used anywhere else

I think this module should be removed and the function defined where they are used.