cta-observatory/ctapipe #1114: suggestions for dependency cleanup

kosack, 2019-07-24:
In an effort to automate the conda package build, I'm now using conda-build's jijja2 templating to extract the list of ctapipe dependencies from setup.py and transfer them automatically to the build recipe.  Right now this results in a recipe with the following build requirements.  Do these look reasonable and not too strict? I've put a few comments here as well:

```yaml
        - pandas
        - numpy >=1.11.3,<2.0a0
        - scikit-learn   # should we fix a version here?
        - astropy~=3.0
        - matplotlib>=2.0   # current is 3.0, do we use any features not in 2.0?
        - scipy>=0.19 . # current is 1.3.0, so perhaps should at least increase minimum to ~=1.0
        - bokeh>=1.0.1 # do we need the minor version numner here? why not ~=1.0
        - iminuit # current is 1.3.7, increase to >=1.3?
        - psutil # only used to get some CPU info in provenance, could maybe 
        - python >=3.7,<3.8.0a0
        - tqdm # should perhaps make this optional?
        - numba # need a min version? Maybe ok to allow any due to rapid development
        - pytables # should req at least >3, since large API change from 2 to 3
        - eventio~=0.20.1 . # should *not* pin this too strictly, since may have features added
        - traitlets # should be >=4.1 (or ~=4.1) , since API changed. 
```

Note that we perhaps should be more consistent with how we pin things (`>' vs `>=x,<=y` vs `~=X`)

Probably the best thing is to use the *compatible version* marker `~=` when possible, since this is intended to match a version with no API change, and each package is supposed to specify what numbering scheme it is using (e.g. semantic, DVCS, Olson, ...)

kosack, 2019-07-24:
I guess the biggest question is how to pin *eventio*, since last time it created problems by being too strict.  In Prod4 and above, we may get new data structures that need support, but don't break anything backward, so I would suggest at least specifying a minimum version, or compatible version (though not sure what numbering scheme it uses).

MaxNoe, 2019-07-24:
> though not sure what numbering scheme it uses

Same as ctapipe. Last number for compatible changes, second number for incompatible, first is 0.
We could call it 1 and start using the real SemVer.

I think we already implemented everything needed for Prod4, I spoke to Konrad.
See this issue https://github.com/cta-observatory/pyeventio/issues/179

Somewhere here the wish for a more complete telescope / optics description was raised, that would introduce a new object type but that should be able to be backwards compatible.

MaxNoe, 2019-07-24:
I also feel that bokeh should probably be optional.

kosack, 2019-07-24:
> Maybe the bokeh stuff should probably be optional?

yes, it's only needed for the event-display gui thing, which could just give a warning like "please install Bokeh" if it's not there.

kosack, 2019-07-24:
> Same as ctapipe. Last number for compatible changes, second number for incompatible, first is 0.
We could call it 1 and start using the real SemVer.

Ok, so we could just restrict it to either `>0.22.0,<0.23.0` for now, or you can move to 1.0 if you are ready.

kosack, 2019-08-20:
fixed in #1118

closed 2019-08-20 by kosack