cta-observatory/ctapipe #1116: writing string with HDF5TableWriter

vuillaut, 2019-08-05:
How can I write string fields with `HDF5TableWriter`?

```
In [1]: from ctapipe.io.containers import Container, Field
   ...: from ctapipe.io import HDF5TableReader, HDF5TableWriter
   ...:
   ...: class TextContainer(Container):
   ...:     text = Field("hello", "some text")
   ...:     number = Field(int, "some number")
   ...:
   ...: text_container = TextContainer()
   ...: text_container.text = 'bonjour'
   ...: text_container.number = 2
   ...:
   ...: with HDF5TableWriter('testfile.h5', group_name='data') as writer:
   ...:     writer.write('text', [text_container])
   ...:
   ...: with HDF5TableReader('testfile.h5') as reader:
   ...:     tab = reader.read('/data/text', TextContainer())
   ...:     new_text_container = next(tab)
   ...:
   ...: new_text_container.text
Table '/data/text' is missing column 'text' that is in container TextContainer. It will be skipped.
Out[1]: 'hello'
```

kosack, 2019-08-05:
Can you give a use case? I can't really image a place where you want to write strings to the binary file, other than in headers. You can use an ENUM to convert most strings to integer keys. 

The answer so far though is that strings are not currently supported but we could add it.  I would caution against allowing variable-length strings, though, since that causes a large speed decrease since they can't be stored contiguously with the rows.

kosack, 2019-08-05:
see for example: https://cta-observatory.github.io/ctapipe/examples/containers_with_enums_and_table_writer.html

vuillaut, 2019-08-05:
Yes actually I might be trying to write information at the wrong place... 

The use case is basically provenance: writing info about used source files, algorithms, etc...
Note that I also would like to be able to merge files and thus a table seemed like a good idea, so I can keep track of each line for each file.

vuillaut, 2019-08-05:
Note2: in my first implementation, I went for `root._v_attrs` but then I'm not leveraging on HDF5TableWriter and Containers and it I have to add some custom code.

kosack, 2019-08-05:
Those should be in the metadata, any `Container` class has a `.meta` attribute which is a dictionary.  Anything you put in there is written to the HDF5 attributes of that dataset

kosack, 2019-08-05:
Note that it is written the first time `writer.write()`  is called, while it is constructing the table (you can only write the headers once), so make sure you define what you want in `container.meta` first before writing any rows out

kosack, 2019-08-05:
Also note that some attributes are written automatically by HDF5TableWriter, such as units and the ctapipe version:

[image]

kosack, 2019-08-06:
> Note2: in my first implementation, I went for root._v_attrs but then I'm not leveraging on HDF5TableWriter and Containers and it I have to add some custom code.

I think that's ok, since `HDF5TableWriter` is not meant to be a general data manager, just the part that deals with a single dataset in a file, it shouldn't get overloaded with too much functionality. We eventually will need to build some helper functions or a class for managing groups of datasets and metadata inside a file.

MaxNoe, 2019-08-26:
The container contains the default value if no value could be read? That is dangerous right?

kosack, 2019-08-30:
> The container contains the default value if no value could be read? That is dangerous right?

I guess that is unrelated to this issue, but still a problem.  Perhaps open another issue.  I guess it's ok if the defaults are nan, or it should write nan or None to the output field in that case.

MaxNoe, 2020-03-19:
PyTables still does not support variable length strings.
So we could only use fixed length strings with e.g. setting a size in `Field`, which would be related to #1224 or use the length of the first element and warn if a string gets truncated.

maxnoe, 2022-01-04:
This is now implemented using a `max_length` either given to the field or from the first element written in #1817

closed 2022-03-29 by maxnoe