cta-observatory/ctapipe #1122: Problem in calling "next" wiht Simtel inputfiles

FrancaCassol, 2019-09-25:
Hi,
if I call two times the "next" function on the same source file I get the error "Only forward seeking possible". Try:

```
from ctapipe.utils.datasets import get_dataset_path
from ctapipe.io import event_source

filename = get_dataset_path("gamma_test_large.simtel.gz")
source= event_source(filename)

# fist call
event = next(iter(source))
print(event.r0.event_id)

# second call
event = next(iter(source))
```
This is annoying if one wants to look slowly at successive events, can perhaps be easily fixed?

By the way, I could also not get the seeker working, the following code gives the error: "Reader is not compatible as input to the event_source is a stream (seeking not possible)"

```
from ctapipe.utils.datasets import get_dataset_path
from ctapipe.io import event_source

filename = get_dataset_path("gamma_test_large.simtel.gz")
source= event_source(filename)

seeker = EventSeeker(source)
event = seeker[1]
```
Should I use an other syntax?

Thanks

MaxNoe, 2019-09-25:
You must only call `iter` once! `iter` generates a new iterator.

So change to:

```
it = iter(source)
event = next(it)
event = next(it)
```

vuillaut, 2019-09-25:
Note: if you want to seek again the same event, you may want to use `back_seekable=True`

FrancaCassol, 2019-09-26:
Thanks very much @vuillaut,
and what about the seeker do you confirm that one can not use it MC data?

vuillaut, 2019-09-26:
> Thanks very much @vuillaut,
> and what about the seeker do you confirm that one can not use it MC data?

Works for me if I use `back_seekable=True`

```
from ctapipe.utils.datasets import get_dataset_path
from ctapipe.io import event_source
from ctapipe.io import EventSeeker
filename = get_dataset_path("gamma_test_large.simtel.gz")
source= event_source(filename, back_seekable=True)
seeker = EventSeeker(source)
event = seeker[1]
```

FrancaCassol, 2019-09-26:
Ah, I miss understood your explanation. Perfect. It works.
Thanks also to @MaxNoe (I realise now that you were two :-) )

closed 2019-09-26 by FrancaCassol