cta-observatory/ctapipe #1127: Scale of the pulse time channel correct?

TjarkMiener, 2019-10-04:
I was modifying the [DL1DH writer](https://github.com/cta-observatory/dl1-data-handler/tree/update_ctapipe) to make it compatible with the newest ctapipe release. As discussed in cta-observatory/dl1-data-handler#74, it seems that the scale of pulse time channel is not correct (not in nanoseconds). However, the charge channel looks good to me. I also tested for FlashCam and DigiCam with the same behaviour. So maybe I'm messing something up with the configurations of the `CameraCalibrater` or the `GainSelector`?

watsonjj, 2019-10-22:
The pulse time is currently extracted in units of the sampling rate. If the sampling rate is per ns, then the units are ns.

The method in `extract_pulse_time_around_peak` extracts the weighted average of the samples around the peak. This can result in extreme values for pixels with no signal. There is no limit applied to the return value. Perhaps we wish to constrain the value to be within the readout window. Or perhaps we want to consider a more robust method.

The time image should look logical after tailcut cleaning. Or the pulse time can be constrained in the analysis.

Finally, the pulse time does need to be converted to ns for cameras that don't sample per ns. I will create a PR for this.

MaxNoe, 2019-10-22:
> This can result in extreme values for pixels with no signal. 

Why? A weighted average should never produce values outside min / max of the samples. So how can this produce values outside of the readout window?

watsonjj, 2019-10-23:
> A weighted average should never produce values outside min / max of the samples

Sure it can, if the values being averaged seem to be decreasing or increasing (almost like they were on the edge of a pulse) then the average will be outside the range. If these values are not on the edge of the pulse, and are just a symptom of waveform noise/variations, then the resulting weighted average can be way outside the range.

```
import numpy as np
from matplotlib import pyplot as plt

sample_i = np.array([17, 18, 19, 20, 21, 22])
value = np.array([3.970581  ,  0.22058105,  1.470581  ,  3.0881958 , -3.8970947 , -4.779419])
plt.plot(sample_i, value)
print(np.average(sample_i, weights=value))
### -351.5269059715945
```

[image]

vuillaut, 2019-10-23:
> The method in `extract_pulse_time_around_peak` extracts the weighted average of the samples around the peak. This can result in extreme values for pixels with no signal. There is no limit applied to [...]

As a first fix, I'd recommend limiting the time values to the readout window.

> The time image should look logical after tailcut cleaning. Or the pulse time can be constrained in the analysis.
> 

Not all methods (tailcut) clean the image ;-)

MaxNoe, 2019-10-23:
> Sure it can, if the values being averaged seem to be decreasing or increasing 

Oh, you are accepting negative weights? That's not well-defined at all. If by accident you have weights summing up to 0, this will even raise an error.

watsonjj, 2019-10-23:
> Oh, you are accepting negative weights

Sounds like not accepting them is a logical solution

MaxNoe, 2019-10-23:
> As a first fix, I'd recommend limiting the time values to the readout window.

No! These are invalid values and are a strong sign that something is wrong (e.g. no signal at all in the pixel). Never replace invalid values by a valid number. 

> Sounds like not accepting them is a logical solution

This could also lead to problems (a single noisy negative sample in a otherwise nice pulse). Not that easy.

For FACT we fit a third order polynomial to the rising edge and use it's inflection point as arrival time. robust but slow...

watsonjj, 2019-10-23:
> This could also lead to problems (a single noisy negative sample in a otherwise nice pulse). Not that easy.

But this solution should still be better than what we have currently.

MaxNoe, 2020-03-19:
fixed by #1143

closed 2020-03-19 by MaxNoe