cta-observatory/ctapipe #1130: What's the current way to use EventSeeker?

STSpencer, 2019-10-10:
Hi,
I'm having a bit of trouble with the EventSeeker classes, examples and the current documentation give conflicting reports about how the current version of ctapipe uses them. Using the code here :https://cta-observatory.github.io/ctapipe/api/ctapipe.io.EventSeeker.html#ctapipe.io.EventSeeker I get errors related to using 'reader' or 'event_source' as the EventSeeker argument, or errors along the lines of : OSError: Reader is not compatible as input to the event_source is a stream (seeking not possible) . Does anyone know what the current correct way to select a slice of events is?
Thanks a lot

kosack, 2019-10-12:
I think you have to enable the "back_seekable=True" option when creating the EventSource, by default it assumes it's a stream.

STSpencer, 2019-10-14:
Thanks a lot!

closed 2019-10-18 by STSpencer