cta-observatory/ctapipe #1132: Coordinate frame transformations speed

watsonjj, 2019-10-14:
The speed of converting between AltAz and ICRS with astropy coordinates is often one of the primary bottlenecks for our analysis.

Improvements to address this has been discussed in astropy PR: https://github.com/astropy/astropy/pull/6068

However progress appears to be stalled.

It can be argued that the flexibility and accuracy provided by astropy coordinates is way beyond what we require in ctapipe, and therefore we could consider our own implementation (in C or numba for example).

We should discuss how to proceed

kosack, 2019-10-21:
- right now we don't do AltAz to ICRS, except for star position in the camera (so maybe no need to optimize)
- we do perform many "local" transforms (CameraFrame, TiltedFrame, HorizonFrame, GroundFrame) that are not optimized

Let's look into speeding some of those up

Also, remove from HillasReconstruction places where we construct a Frame for each event.

maxnoe, 2020-10-26:
The corresponding PRs are now merged in astropy and will be released with astropy 4.2, which is now in feature freeze so should be out in a couple of weeks.

The interpolation based approach speeds up transformation between AltAz / ICRS around a factor of 100 without sacrificing any precision (still at the < micro arcsecond even when interpolating on the minute time scale).

Since ctapipe right now does not use any ICRS <-> AltAz transformations, I will close this. 

When we add things like this, let's remember to use the interpolating appraoch if we do require AltAz / ICRS conversions for large arrays of coordinates.

See https://docs.astropy.org/en/latest/coordinates/index.html#improving-performance-for-arrays-of-obstime for how to use this approach.

closed 2020-10-26 by maxnoe