cta-observatory/ctapipe #1135: How many events in a simtel  file?

FrancaCassol, 2019-10-17:
Hi,
is it possible to know how many events are inside a simtel file without making a loop over all of them?

Thanks

MaxNoe, 2019-10-17:
Do you mean simulated shower events or triggered telescope events?

The first is possible to know from the run header and simulation header. The latter is only possible by looping through the file as far as I know.

If you are only interested in the numbers and not the actual data, it might be faster just to use `eventio.EventIOFile` and look for the Histograms at the end of the file.

MaxNoe, 2019-10-17:
Like this:
```
from eventio import EventIOFile, Histograms
from eventio.search_utils import yield_toplevel_of_type

with EventIOFile(inputfile) as f:
    for o in yield_toplevel_of_type(f, Histograms):
        hists = o.parse()
```

FrancaCassol, 2019-10-17:
I see, and if I open the file with the event_source?

```
from ctapipe.io import event_source
source = event_source(data, max_events=None, back_seekable=True)
```

FrancaCassol, 2019-10-17:
Actually I do that so to have a code MC and real data compatible

MaxNoe, 2019-10-17:
You can access `source.file_.histograms` **after** looping through all events

FrancaCassol, 2019-10-17:
But I would like to have the info when I open the file :-(

MaxNoe, 2019-10-17:
That's just not availabe for the simulations!

The CORSIKA output is directly piped into simtel array which processes each event individually and writes it out to the file.

The information is only available, after the simulation run has finished and is thus written at the very and of the file.

The histograms are the last object written to the files.

FrancaCassol, 2019-10-17:
This is a problem for data formats without random access, very impractical.

MaxNoe, 2019-10-17:
But it's huge advantage when writing the simulation software and streaming this data! 

That's always a tradeoff and I understand why the decision was made the way it was.

kosack, 2019-10-17:
by the way, just to note, This will no longer be a problem once I've merged in all my changes for the DL1 format (I'm still waiting for the preliminary PRs to be accepted before doing the big ones though!) . With the DL1 files in HDF5 format, you can easily get the thrown shower histograms, and the number of recorded showers without having to seek.

In simtel, the main problem is that even though the number of *simulated showers* is known at the start, the number of those detected and read out is not known, and of course the process may be terminated early such that even the original thrown showers is not correct.

kosack, 2019-10-17:
Can you tell me the use case for knowing this when opening the simtel file? Is it just to know how long it will take to process or something?  If it's important, we can keep it in mind for the final DL0 format, which probably won't be simtel)

FrancaCassol, 2019-10-17:
I use this information in the calibration tool for cross-checking the asked event statistics, in my case I can indeed survive without this information, but I think  it is in general very useful at the opening of a data file to know how many events it contains. 

An other important feature would be to have the possibility to jump to specific events without looping on all the previous ones

MaxNoe, 2019-10-17:
> An other important feature would be to have the possibility to jump to specific events without looping on all the previous ones

Again, as this is a common wish here: this is just not possible with the majority of files that can be efficiently written to disk!

Random Access files normally cannot be streamed and vice versa!

FrancaCassol, 2019-10-17:
One could not create in parallel to the data file an index  file with the event position and use it to jump directly to specifics events afterwards?

MaxNoe, 2019-10-17:
Yes you could, but that would have to live in a separate file so you'd need to always handle two files.

Or you would need to change the header of the file after it. But these are both methods that immensely complicate the file handling on the writer side and increases risks of data corruption.

All this for a relative small benefit. In normal event processing, you never need to go to specific events or to know the exact number of events beforehand.

This is a very basic computer science problem without good solutions. You have to make these trade-offs.

Simpler file-formats that are streamable vs. Random Access and Metadata in a header.

MaxNoe, 2019-10-17:
And, what I completely forgot: compression destroys this basically completely. With compressions like gzip or zstd, you always need to start at the front of the file.

Also eventio files are heavily context dependent. You cannot just skip to a telescope event. You would need to read the last Monitoring data, Shower Information, Array Event, Trigger Information that all live in multiple containers before that Telescope event.

So, sorry, but I think it's just not feasable.

The FACT zfits format tries to be a middle ground, but that comes with a pretty high complexity, even more with the CTA protobuffer filled zfits.
And is not standard fits conform anymore.

kosack, 2019-10-18:
In any case, I think in the end we'll have a database where we can look up metadata, which could easily include the number of events. Reading it from file is not necessary, except as a hack now.

vuillaut, 2019-10-18:
(same goes for HDF5 DL0 ;-) )

MaxNoe, 2020-03-20:
I'll close this, as I think the discussion has probably answered all questions and we have little control over this.

Feel free to reopen if there are any issues remaining.

closed 2020-03-20 by MaxNoe