cta-observatory/ctapipe #1140: telescope naming (revisit)

kosack, 2019-10-21:
- Telescope names are used for naming datasets in the DL1/DL2 output files.  
- Right now, we name these using  "opticstype_opticssubtype_camera" naming, but for telescopes like "MST", there is no subtype, so we use "MST_MST_NectarCam", which is a bit ugly (it's nicer for something like "SST_ASTRI_CHEC"). Can we do better?  
- should be parsiable (e.g. easy to turn a name back into a telescope with correct optics type and camera type)
- should we encode versions? (NectarCam_002)l or just leave that to metadata?
- the problem is less once CTA has chosen which telescopes it will build, so we have to support fewer combinations.

kosack, 2019-10-21:
Discussion: just combine optics and camera into a single "type", so we just use "telescope-class_telescopename" (e.g. `MST_NectarCam`, or `SST_ASTRICHEC`) 

- should consider future where SSTs/LSTs/MSTs are relatively fixed. 
- is there a real name for LSTCam?
- should be future proof: if we change MST structure, or swap cameras....

MaxNoe, 2019-10-21:
I think the LST cam was refered to as `DragonCam` in the past

kosack, 2020-01-16:
I also just noticed that SCTs are not treated correctly.  The Telescope Class should only be MST, SST, or LST, but for SCT cameras we get a telescope name of `SCT_SCT_SCTCam` instead of `MST_SCT_SCTCam`. 

So I still think having the three things is best (which is how it worked before it was restructured): 
- tel class: LST, MST, SST
- tel type: 
  - LST: XXX (only 1 so far)
  - MST: XXX or SCT (where XXX is what? "Standard", "DC", "Classic"?)
  - SST: 1M, ASTRI, GCT
- camera type
  - LSTCam
  - NectarCam
  - FlashCam
  - etc.

Currently this is used for some statistics in the analysis, e.g. `n_mst` is the number of MSTs that triggered (regardless of which type)

maxnoe, 2022-07-12:
The simtel metadata has OPTICS_NAME and CAMERA_NAME, which I think is good.
`
I would go for having the telescope type be `{OPTICS_NAME}_{CAMERA_NAME}`, e.g. `LST_LSTCam` / `MST_NectarCam` / `ASTRI_CHEC`, type should be one of LST / MST / SST, but this could be purely determined based on mirror area / diameter.

Then, `TelescopeDescription` should also have a name, which could be `LST-1`, `MST-02` for each telescope, but is not participating in the hash (or we need to seperate `TelescopeType` from `Telescope`.