cta-observatory/ctapipe #1159: Proposal: require "black" code formatter as pre-commit hook

kosack, 2019-10-28:
To avoid style change problems, I suggest we require devs to either manually run "black" (or in their editor of choice), or install it as a pre-commit hook (see e.g. https://ljvmiranda921.github.io/notebook/2018/06/21/precommits-using-black-and-flake8/)

Black Documentation: https://black.readthedocs.io/en/stable/index.html

For bits of code where black really does a worse job (are there any?), I think it can be disabled explicitly similar to yapf's `# yapf: disable`

watsonjj, 2019-10-28:
Do we also adopt the 88 character line length?

kosack, 2019-10-28:
It's ok with me - I already have the tests only check for 90 I think, even though we recommend 80.  The idea of 88 is just that you should normally stick to 80, but it's ok to go over a bit in cases where it makes sense.  The alternative is to specify it in a config file, so black uses 80.

vuillaut, 2019-11-22:
To be considered:
https://github.com/marketplace/actions/black-code-formatter

kosack, 2019-11-22:
How does that work? Does it modify the code when you push? Wouldn't that create conflicts? Or does it just warn that the user didn't run black?

kosack, 2019-11-22:
Ah I see it just uses the --check option, so might be a good idea ! I guess it's like a harsher version of what codacy is already doing with its pep8 checking, but also applying to actual formatting.

MaxNoe, 2020-05-29:
done

closed 2020-05-29 by MaxNoe