cta-observatory/ctapipe #1161: Builds on Travis not building correctly

garciagenrique, 2019-10-29:
Hi all, 

Making some pull requests on another repository, we have realised (thanks @devine73 and @jacquemier) that travis is not building correctly. 
We have also seen, looking into travis' log, that they are other pull requests that aren't building correctly either (PRs by @FrancaCassol in ctapipe_io_lst and ctapipe_io_nectarcam). 

In all the cases we are finding an error in common: 
`error in ctapipe-extra setup command: "values of 'package_data' dict" must be a list of strings (got '*')` that might come from the setup configuration within the ctapipe-extra repository.

Any ideas / way to solve this ?

Thanks in advance

MaxNoe, 2019-10-29:
This is an issue for for ctapipe-extra, not ctapipe. It was already fixed in https://github.com/cta-observatory/ctapipe-extra/pull/27

You could also downgrade to a older version of `setuptools`, that did not raise this error. 

We should make a new release of ctapipe-extra, but this would also need adaption of ctapipe-tests.

@kosack maybe we should restore the mc `NectarCam`, that created the problems i had when using the master of ctapipe-extra

MaxNoe, 2019-10-30:
Can you point to a travis build that has the problem?