cta-observatory/ctapipe #1177: can't instantiate abstract class EventSource

SeiyaNozaki, 2019-11-19:
I got the following error when I executed EventSource.

```
>>> from ctapipe.io import EventSource
>>> eventsource=EventSource("~/Work/CTA/Realdata/LST-1.1.Run01575.0000.fits.fz")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/nozaki/Install/anaconda3/envs/cta-dev/lib/python3.7/site-packages/traitlets/traitlets.py", line 955, in __new__
    inst = new_meth(cls)
TypeError: Can't instantiate abstract class EventSource with abstract methods _generator, is_compatible, subarray
```

I think it came from that there is no codes in those definition(_generator, is_compatible, subarray) of io/eventsource.py.
But I'm not sure what is expected for those functions...
(I found this error because I got similar error when I executed LSTEventSource on cta-lstchain.)

Please have a look?

sizun, 2019-11-19:
Did you try:

```
from ctapipe.io import event_source
eventsource = event_source("~/Work/CTA/Realdata/LST-1.1.Run01575.0000.fits.fz")
```

dneise, 2019-11-19:
Hello Seiya, thanks for reporting. 
I think @sizun is right. You might also want to have a look at this tutorial again:

https://cta-observatory.github.io/ctapipe/tutorials/ctapipe_handson.html#Part-1:-load-and-loop-over-data

@cta-observatory/cta-devs do you think one could remove `EventSource` from [`io/__init__.py`](https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/io/__init__.py#L3) so that `EventSource` is not part of the public interface anymore?

kosack, 2019-11-19:
You can also instance it using the factory method `EventSource.from_url()` . The base class is not meant to be instantiated, rather one of the factory methods should be used to return an automatically selected sub-class based on the input.

kosack, 2019-11-19:
> @cta-observatory/cta-devs do you think one could remove EventSource from io/__init__.py so that EventSource is not part of the public interface anymore?

No, it is part of the public interface: the EventSource base class allows event sources to be configured in a Tool.  The `event_source()` helper function is just for interactive use, and shouldn't be used in a Tool (since it will not be properly configured from a config file )

kosack, 2019-11-19:
For examples of using the EventSource, look in `ctapipe/tools/*.py`, and for a really detailed example, see #554

kosack, 2019-11-19:
The idea, is that you should never instantiate a subclass (like LSTEventSource) yourself - those should not be public. Rather, you should use one of the factory methods in EventSource itself to get what you want (either: `EventSource.from_url( filename )` → autodetect type from input file, `.from_config(config)`: get type from user configuration dictionary).  The former is preferred, since normally you want your tool to support any input based on its file type.

pawel21, 2019-11-20:
Sorry, but when I executed:
`event_source("LST-1.1.Run01531.0000.fits.fz")`

I got following error
```
ValueError: Cannot find compatible EventSource for 
	url:LST-1.1.Run01531.0000.fits.fz
in available EventSources:
	['SimTelEventSource']
```
The same `EventSource.from_url( filename )`

MaxNoe, 2019-11-20:
Do you have `ctapipe_io_lst` installed? Looks like you do not. `conda list` or `pip freeze`

pawel21, 2019-11-20:
Hi @MaxNoe 
Thank you,
I think, I have
`conda list` gave me:

```
...
coverage                  4.5.4            py37h516909a_0    conda-forge
cryptography              2.8              py37h72c5cf5_0    conda-forge
ctapipe                   0.6.2            py37ha8d69ae_2    cta-observatory
ctapipe                   0.5.3.post505             <pip>
ctapipe-extra             0.2.14               h1f2461a_0    cta-observatory
ctapipe-io-lst            0.2                       <pip>
cycler                    0.10.0                     py_2    conda-forge
cython                    0.29.14          py37he1b5a44_0    conda-forge

...
```

MaxNoe, 2019-11-20:
That ctapipe version looks rather old, there was a 0.6.1, 0.6.2 and 0.7 release since 0.5.3. Also you seem to have two ctapipe instalations. That's weird.

Try
```
pip uninstall ctapipe
conda uninstall ctapipe
```

And then reinstall.

kosack, 2019-11-20:
I think you should be using at least ctapipe=0.7, since the older ones were not compatible with LST data.  You can install it via `conda install -c cta-observatory -c conda-forge ctapipe=0.7` (but as Max said, uninstall the old ones first, since it's strange to have 2 installed at once). Or better yet, create a new environment (follow the developer guide instructions, but then install ctapipe via conda)

pawel21, 2019-11-20:
When I tried `conda install -c cta-observatory -c conda-forge ctapipe=0.7`
I got the following output:

```
Fetching package metadata ...............
Solving package specifications: 

PackageNotFoundError: Packages missing in current channels:
            
  - ctapipe 0.7* -> astropy ~=3.0
  - ctapipe 0.7* -> bokeh ~=1.0
  - ctapipe 0.7* -> matplotlib ~=3.0
  - ctapipe 0.7* -> pytables ~=3.4
  - ctapipe 0.7* -> scipy ~=1.2

We have searched for the packages in the following channels:
            
  - https://conda.anaconda.org/cta-observatory/linux-64
  - https://conda.anaconda.org/cta-observatory/noarch
  - https://conda.anaconda.org/conda-forge/linux-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://repo.continuum.io/pkgs/main/linux-64
  - https://repo.continuum.io/pkgs/main/noarch
  - https://repo.continuum.io/pkgs/free/linux-64
  - https://repo.continuum.io/pkgs/free/noarch
  - https://repo.continuum.io/pkgs/r/linux-64
  - https://repo.continuum.io/pkgs/r/noarch
  - https://repo.continuum.io/pkgs/pro/linux-64
  - https://repo.continuum.io/pkgs/pro/noarch
```            
@SeiyaNozaki are you able to read data with new version of ctapipe?
(Maybe problem is with my env/computer)

MaxNoe, 2019-11-20:
The default channel seems to be missing?

MaxNoe, 2020-03-19:
I think this is an old installation related issue, so I'll close this.

Feel free to reopen, if you still have issues.

closed 2020-03-19 by MaxNoe