cta-observatory/ctapipe #1178: Not understanding coordinate transformations

dneise, 2019-11-21:
What is the easiest way to transform distances in the camera plane (given in meters) to distances in the sky in degrees, without actually solving the geometry of the problem again. 

In small angle approx this is of course just dividing any distance by the equivalent focal length and calling it a day, but we know of course that our telescopes are not really thin lenses .. so this approximation fails in certain cases. 

I assumed there was a library function, which performs these transformations within a certain level of precision, but it seems not to be the case, right?

I assume the deviations from the thin lense approximation have a larger effect than refraction of the atmosphere, which is taken into account already by astropy.  right?

MaxNoe, 2019-11-21:
What you are talking about is the angular distance between two points in the `TelescopeFrame`,
so transform the two points in CameraFrame to TelescopeFrame and then get the angular distance.

```python
from ctapipe.coordinates import CameraFrame, TelescopeFrame
from astropy.coordinates import SkyCoordInfo
import astropy.units as u

camera_frame = CameraFrame(focal_length=38 * u.m)
telescope_frame = TelescopeFrame()

 c = SkyCoord(x=[0 * u.m, 1 *u.m], y=[0 * u.m, 1 * u.m], frame=camera_frame)
 t = c.transform_to(telescope_frame)
print(t[0].separation(t[1]))
```

MaxNoe, 2019-11-21:
If you want to take refraction into account and all this, you probably have to go the full way from `CameraFrame` to `ICRS` (ra, dec), which requires `obstime`, `location`, `telescope_pointing` and `focal_length` to be set on the coordinate.

MaxNoe, 2019-11-21:
However, this would do a precision that is almost certainly overkil for anything IACT related.

dneise, 2019-11-21:
I totally agree. Thanks!

dneise, 2019-11-21:
I just wanted to mention, this is about 260x slower than just dividing by the focal length ... I'm sure you know it ... but ... for some use cases this might potentially be an issue ...

closed 2019-11-21 by dneise