cta-observatory/ctapipe #1180: add mc true image sum

kosack, 2019-11-21:
In MCs without the `mc_photo_electron_image` (the true number of PEs on the photosensor plane without NSB noise), it seems there may be at least the sum of this image stored in the MC data:

From @moralejo: 
```
mc_pe_sum = mc_event.mc_pesum.num_pe[itel]
```

We should read this in the SimTelEventSource, and put it in the `MCCameraEventContainer`

MaxNoe, 2019-11-21:
I think you are referring to this simtel object 

https://github.com/cta-observatory/pyeventio/blob/84a7dc15b85506216d30204bbbb2561dc61720d6/eventio/simtel/objects.py#L1594

(Type 2026)

which is in the event dictionary of the `eventio.SimTelFile` as `photoelectron_sums` which is currently not used in simtel evnet source.

kosack, 2019-11-21:
@MaxNoe  yes, that appears to be it.   Not sure if it exists in all simulations, but if so, it's worth it, since we may not store the full image in all sims to reduce space (even though I would suggest keeping it in any test/train datasets)

MaxNoe, 2020-05-01:
I produced a smallish test file with true mc images.

I'll open a PR for ctapipe-extra

MaxNoe, 2020-05-18:
Since the file is now in extra and we updated the version here, we could use this.

This is the content of the mc pe sums container (for the four LST file):

```
n [15]: e['photoelectron_sums']                                                                                                              
Out[15]: 
{'event': 100,
 'shower_num': 1,
 'n_tel': 4,
 'n_pe': array([2623,  227,  775, 2824], dtype=int32),
 'n_pixels': array([0, 0, 0, 0], dtype=int32),
 'pixel_pe': [],
 'photons': array([44231.836, 48354.355, 50571.93 , 40675.74 ], dtype=float32),
 'photons_atm': array([25795.885, 27785.705, 29079.496, 24070.93 ], dtype=float32),
 'photons_atm_3_6': array([17677.818, 18906.129, 19839.078, 16423.305], dtype=float32),
 'photons_atm_qe': array([4583.637 , 4890.8315, 5126.1567, 4277.5977], dtype=float32),
 'photons_atm_400': array([6928.6274, 7369.2896, 7772.921 , 6387.361 ], dtype=float32)}
```

commit 300a40a8 (2022-01-28): Read true_image_sum, fixes #1180
commit 4edf54a5 (2022-01-28): Merge pull request #1814 from cta-observatory/true_pe_sum

Read true_image_sum, fixes #1180
closed 2022-01-28 by maxnoe