cta-observatory/ctapipe #1182: Can Containers get properties?

dneise, 2019-11-22:
Is it possible to give properties to container classes? So a container can be conveniently asked for stuff, that can be calculated from its contents, without having to repeat this calculation wherever it is needed? 

I assume we could cache the result, in order to improve performance if the computation was somehow complicated, I assume, while still keeping the memory footprint of a stored container on disk as small as needed.

kosack, 2019-11-22:
We didn't want them to turn into analysis classes, but I think for simple properties it's probably ok.  I just wanted to avoid the usual problem of data classes accumulating a ton of functions, when they are really supposed to just be ORM-like structs.

kosack, 2019-11-22:
I guess those properties would not get written to disk as a column? that would be more complex to implement.

kosack, 2019-11-22:
It certainly "works" with the current implementation, so just a question of if it is useful:
```py3
In [13]: class C(Container):
    ...:     x = Field(0,"test")
    ...:     @property
    ...:     def x2(self):
    ...:         return self.x**2

In [14]: c = C()

In [15]: c
Out[15]:
__main__.C:
                             x: test

In [16]: c.x2
Out[16]: 0

In [17]: c.x=12

In [18]: c.x2
Out[18]: 144
```

Note that x2 would not get written to disk, since it's not a Field, so it doesn't increase the storage footprint.

dneise, 2019-12-03:
So we found an example here:
https://github.com/cta-observatory/ctapipe/blob/8a1ac929db346c74d8380e2ea765ba7965d6bad5/ctapipe/io/containers.py#L575-L579

where `phi` and `inclination` are completely dependent on `center_x` and `center_y`. I thought it would be good to make them into properties and cache them, but @MaxNoe correctly pointed out that the cache needs to be invalidated once attributes of the container get modified. 

Not caching them, is also an option, much simpler.

Another point is .. if phi and incl depend on x and y .. the same is true the other way around .. what if somebody wants to specify phi and incl and have x and y calculated for them ...

dneise, 2019-12-12:
I think this is in general not well likes in the community, I will close this issue and nver talk about it again :-D

closed 2019-12-12 by dneise

MaxNoe, 2019-12-12:
How did you get this impression?