cta-observatory/ctapipe #1183: GroundFrame directions

nbiederbeck, 2019-11-22:
What are the x-, y-, (z-) directions of `ctapipe.coordinates.GroundFrame`?

In MAGIC it is
```
x = north
y = west
```

what about CTApipe?

thomasgas, 2019-11-22:
In ctapipe it is the same as in MAGIC

kosack, 2019-11-22:
and I think so far Z is meters above the array center (not meters above sea-level)

thomasgas, 2019-11-22:
> and I think so far Z is meters above the array center (not meters above sea-level)

z is in meters above the array observation level which is saved in ``event.mcheader.prod_site_alt``

MaxNoe, 2020-03-19:
It should be noted that this is different from the cartesian representation of astropy AltAz coordinates.

This could bite us, right?

MaxNoe, 2020-03-19:
```
In [3]: AltAz(alt='0d', az='0d').cartesian                                                                                                    
Out[3]: 
<CartesianRepresentation (x, y, z) [dimensionless]
    (1., 0., 0.)>

In [4]: AltAz(alt='90d', az='0d').cartesian                                                                                                   
Out[4]: 
<CartesianRepresentation (x, y, z) [dimensionless]
    (6.123234e-17, 0., 1.)>

In [5]: AltAz(alt='0d', az='90d').cartesian                                                                                                   
Out[5]: 
<CartesianRepresentation (x, y, z) [dimensionless]
    (6.123234e-17, 1., 0.)>

```

So in astropy 0d az = N = x, 90d az = E = y

maxnoe, 2023-01-04:
This is now documented in the `GroundFrame` docstring and we added `NorthingEastingFrame` for easier map plotting.

I think this can be closed.

closed 2023-01-04 by maxnoe