cta-observatory/ctapipe #1184: chaining components that need a final step after all events

kosack, 2019-11-22:
## Algorithms with event-wise and final calls:

I'd like to come back to the issue of how to chain together a sequence of Components in a Tool.

I'm refactoring #1066 to break it into Components, and I need to make a `DL1EventWriter` component, that writes some final tables when all event writing is done.   The general data flow is roughly like this:

```py3
source = event_source(filename)
subarray = source.subarray

calibrate = CameraCalibrator(..., subarray=subarray)
process_images = ImageProcessor(..., subarray=subarray)
write_dl1 = DL1Writer(..., subarray=subarray)

for event in source:
     calibrate(event)
     process_images(event)
     write_dl1(event)

write_dl1.finish()  # easy to forget to call this...
```

## options for improvement

### use context management

```py3
with DL1Writer(...) as write_dl1:
   for event in source:
      calibrate(event)
      process_images(event)
      write_dl1(event)

# cleanup happens automatically at end of "with" block
```

I like this the best if there are only a small number of components with context, but it may get ugly if there are many. Note that python does allow multiple context managers on a line:
```py3
with X() as x , Y() as y, Z() as z:
     ...
```
or there is the `contextlib.ExitStack` class, which is a bit uglier, but works with an arbitrary list of objects (https://docs.python.org/3/library/contextlib.html#contextlib.ExitStack)

### use processor model

A less pythonic alternative would be to use   a *Processor* like interface  (#591) for at least some Components (with minimally `process()` and `finish()` methods, but we could also have an `on_first_event()` method)  That could be done in a destructor ( `__del__`), but then it is called only when the garbage collector decides to call it, so it would likely cause problems. 

As mentioned in #591, one way to go would be to use function composition, chaining , or similar to achieve this. 
```py3
steps = [calibrate, process_images, write_dl1]
map( apply_steps(steps), source )
finish_steps(steps)
```
Note that something similar can still be used in the context management way of doing things.

Right now I'm leaning toward the former (context management) method, but I'm open to suggestions.

kosack, 2019-11-22:
It occurs to be that in the context management option, we can even implement some delayed initialization using `__enter__()`.  So we can just have configuration stuff in `__init__`, and longer tasks like file handling inside `__enter__`

```py3
In [74]: class C(contextlib.AbstractContextManager):
    ...:     def __init__(self,):
    ...:          print("INIT C")
    ...:     def __enter__(self, ):
    ...:          print("ENTER C")
    ...:     def __exit__(self, type, value, traceback):
    ...:          print("EXIT C", type, value, traceback)
    ...:     def __call__(self, x):
    ...:         print("CALLED C with", x)
    ...:
    ...: calibrate = C()
    ...:
    ...: print("Start event loop")
    ...:
    ...: with calibrator:
    ...:     for ii in range(3):
    ...:         calibrate(ii)
    ...:
    ...:
    ...:
```
```
INIT C
Start event loop
ENTER C
CALLED C with 0
CALLED C with 1
CALLED C with 2
EXIT C None None None
```

kosack, 2019-11-22:
I think my main question now is whether or not to force a particular set of abstract methods on all Components, to avoid that the user has to implement `__call__`, `__enter__`, and `__exit__` directly.  We could define for example:
```py3
class ProcessorComponent(Component):
    @abstractmethod
    def _start()
        pass
    @abstractmethod
    def _finish()
        pass
    @abstractmethod
    def _on_first_event(event)
        pass
    @abstractmethod
    def _process(event):
        pass

    def __enter__(self):
        self._start()
    def __exit__(self, t, v, tb):
        self._finish()
    def __call__(self, event):
        if event.count == 0:
           self._on_first_event(event)
        self._process(event)
 ```

But that may be overkill, and hides what is actually happening perhaps.

MaxNoe, 2019-11-22:
This is something all data streaming frameworks provide. 

See also #591 where large parts of this is discussed

We probably should pick either one or implement it on our own in a way that allows using them. 

E.g. Apache Flink, Apache Storm, thepipe from km3pipe and many, many others.

This is also not a new issue, rather these old issue all go into this direction:

kosack, 2019-11-22:
yes, note that #591 is referenced above already.  This is also how the HESS SASH framework works, but it was found to cause confusion for new users (so lessons learned!), which is why I tried to avoid the "processor" pattern.   But I think the simplest method, i.e. just a context manager, is probably best for now. 

 I think one of the design goals is that it's easy to interface with parallelization systems like `concurrent.futures.ProcessPoolExecutor` or `dask`, so just need to choose something that doesn't break that. Originally I wanted to use generator composition, since that is a really nice way to do it, but it makes it really hard to parallelize later (i.e. `events = write_dl1(process_images(calibrate(source))))`)

kosack, 2019-11-22:
Another example: scikit-learn Pipelines: https://scikit-learn.org/stable/modules/generated/sklearn.pipeline.Pipeline.html

But again, I think sticking to a simple for-loop and context managers for now is not too bad, and quite readable.

vuillaut, 2019-11-22:
I have been trying to use scikit-learn pipelines for ctapipe a while ago but never completed it :-/

PR #1926 (2023-01-10): Tool exit stack
This adds an `ExitStack` to `Tool` which can be used to make sure that e.g. an `EventSource` or a `DataWriter` is properly closed when used by a tool.

Objects that provide context managers should now be used like this in a tool (e.g. `setup`) method:

```
def setup(self):
    self.event_source = self.enter_context(EventSource(parent=self))
```

Fixes #1184