cta-observatory/ctapipe #1186: Plotting events in TelescopeFrame

dneise, 2019-11-25:
What is the easiest way to plot an event in the TelescopeFrame, i.e. not in an x-vs-y frame in meters but in a delta_az-vs-delta_alt frame in degress.

I thought about using [`CameraDisplay`](https://cta-observatory.github.io/ctapipe/api/ctapipe.visualization.CameraDisplay.html?highlight=cameradisplay#ctapipe.visualization.CameraDisplay), but I have the impression, in order to convince the CameraDisplay to plot in the TelescopeFrame, I have to hack the `geom` parameter ... by basically making a deepcopy of it and then modifying the `pix_x` and `pix_y` fields, right? 

I fear, doing this will result in hard to read code. Alternatively I could just create a new `TelescopeFrameDisplay` which is mostly a copy of `CameraDisplay` but plots in the Telescope Frame, but this would copy a lot of code, I assume.

This issue might be connected to #1090, since I assume, if we would calculate the Hillas parameters in TelescopeFrame, we also would like to display them every once in a while, right?

Thanks for your comments.

moralejo, 2019-11-25:
> What is the easiest way to plot an event in the TelescopeFrame, i.e. not
> in an x-vs-y frame in meters but in a delta_az-vs-delta_alt frame in
> degress.
I may be misunderstanding you, but delta_az and delta_alt do not form a
Cartesian system, so for plotting events (shower images?) the xy frame of
the cameras seems to me the natural choice - units can be degrees, and the
axes correspond to great circles intersecting at the fov center.

I thought about using CameraDisplay
> <https://cta-observatory.github.io/ctapipe/api/ctapipe.visualization.CameraDisplay.html?highlight=cameradisplay#ctapipe.visualization.CameraDisplay>,
> but I have the impression, in order to convince [...]

MaxNoe, 2019-11-25:
```
I may be misunderstanding you, but delta_az and delta_alt do not form a
Cartesian system, so for plotting events (shower images?) the xy frame of
the cameras seems to me the natural choice - units can be degrees, and the
axes correspond to great circles intersecting at the fov center.
```
`CameraFrame`  with units in degrees IS the TelescopeFrame. 
If you plot this as x,y values, you basically make a small angle approximation which is pretty good given our field of views (max radius 5 degrees).

> I fear, doing this will result in hard to read code. Alternatively I could just create a new TelescopeFrameDisplay which is mostly a copy of CameraDisplay but plots in the Telescope Frame, but this would copy a lot of code, I assume.

The CameraDisplay already has a `frame` attribute and a `transform_to` attribute. 
I think right now, this only works with `CameraFrame` and `EngeneeringCameraFrame`, as there is stuff like `to_value(u.m)` happening which won't work for TelescopeFrame in degrees.
But, I think this is easily fixable.

dneise, 2019-11-25:
> The CameraDisplay probably also needs some adaptions to be able to handle a CameraGeometry with units in degrees.

okay thanks ... that's nice. I'll have a look into the Display later tonight ...

maxnoe, 2020-10-26:
This is now possible simply by doing:

```
geometry_tel_frame = geometry.transform_to(TelescopeFrame())
display = CameraDisplay(geometry_tel_frame)
```

closed 2020-10-26 by maxnoe

HealthyPear, 2020-10-26:
It is also automatically done from #1408 (in facto part of my solution was inspired by this PR)