cta-observatory/ctapipe #1188: The "miss" parameter à la CTA-MARS

HealthyPear, 2019-11-25:
In the framework of the [comparison](https://forge.in2p3.fr/projects/benchmarks-reference-analysis/wiki/Comparisons_between_pipelines) between _protopipe_ and _CTA-MARS_, and in particular looking at how the latter works in [detail](https://forge.in2p3.fr/projects/step-by-step-reference-mars-analysis/wiki#Direction-Look-up-tables), I encountered a quantity which I think it is not taken into (at least direct) account in _ctapipe_ .
Since I need it for _protopipe_, perhaps it can be useful for others as well.

The quantity I am referring to is called the _miss_ parameter (at least there) and it corresponds to **the minimal distance in degrees between the parametrized image axis and the true source position**.

From `ctapipe.image.hillas.hillas_parameters` many useful quantities are already available and I was thinking about how to implement it (or add it directly to the same method).

Is this already available and I am not seeing it?
If not, did someone already do it somewhere else and we can just migrate it to _ctapipe_ (_lst-chain_ for example)?

watsonjj, 2019-11-25:
Unless I am mistaken, this is also sometimes referred to as "alpha", and is illustrated in this diagram:

[image]

I don't think there is an implementation in ctapipe for this currently. I don't think it should be added to the hillas parameters, and one should probably be careful not to use it in the reconstruction (unless you are doing a source-dependent reconstruction), but we should have an implementation in ctapipe.

This is the implementation I use in the CHEC pipeline:

```python
def calculate_alpha(x_src, y_src, x_cog, y_cog, psi):
    beta = np.arctan2(y_cog - y_src, x_cog - x_src)  # Angle between x and src
    alpha = beta - psi
    alpha = np.arctan2(np.sin(alpha), np.cos(alpha))  # Fix between -pi & pi
    alpha90 = np.abs(np.arctan(np.sin(alpha) / np.cos(alpha)))  # 0 to pi/2
    return alpha, alpha90
```

MaxNoe, 2019-11-25:
Currently there are no source dependent parameters calculated in ctapipe. 

This is something left to higher analysis steps usually, as there might be multiple source in the field of view.

IMHO DL1 to DL3 processing should nowhere depent on a known source position.

MaxNoe, 2019-11-25:
But we could optionally have a `SourceDependentHillasParameters` container, that should also hold the assumed source position. This can then also be produced for Off positions when observing in wobble mode.

HealthyPear, 2019-11-25:
[image]
I used the image uploaded by Jason to clarify a bit.
It related to alpha, but not directly, since you should define a distance along the shower axis to use alpha directly.
I think this quantity is already available using the coordinates of centroid, source position and the angle between shower axis and "x" axis. Am i right?

This is done for MC-based analysis where the source is at the center of the camera, for the moment, but I am not sure one cannot use it also in a real case (the "true" position in the end the one sent to the Drive system).

MaxNoe, 2019-11-25:
You can use `ctapipe.image.hillas.camera_to_shower_coordinates` for this:

```python
_, miss = camera_to_shower_coordinates(source_x, source_y, cog_x, cog_y, psi)
```

moralejo, 2019-11-25:
On Mon, 25 Nov 2019 at 16:29, Maximilian Nöthe <notifications@github.com>
wrote:

> Currently there are no source dependent parameters calculated in ctapipe.
>
> This is something left to higher analysis steps usually, as there might be
> multiple source in the field of view.
>
> IMHO DL1 to DL3 processing should nowhere depent on a known source
> position.
>
I agree. But regardless of where miss is calculated, it can indeed be "used
for a source-independent analysis". In case of MARS, one makes a LUT (with
axes Width / Length and Intensity) by filling in it the average squared
miss. The smaller the value, the more reliable the image axis is. Then one
uses this LUT to assign weights to image axes in stereo reconstruction,
depending only on the Intensity and Width/Length of the image.  Hence in
the "actual analysis" of the events in the "test" sample miss is of course
not used.

A.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

-- 
Avís -
Aviso - Legal Notice - (LOPD) - http://legal.ifae.es 
<http://legal.ifae.es/>

HealthyPear, 2019-11-25:
> You can use `ctapipe.image.hillas.camera_to_shower_coordinates` for this:
> 
> ```python
> from ctapipe.image.hillas import camera_to_shower_coordinates
> disp, miss = camera_to_shower_coordinates(source_x, source_y, cog_x, cog_y, psi)
> 
> ```

Oh it was literally the function above.... :)
Many thanks.

MaxNoe, 2020-03-19:
This is solved, right?

Or do we need this as a function called miss?

closed 2020-03-19 by MaxNoe