cta-observatory/ctapipe #1190: CameraDisplay modifies CameraGeometry during plotting

dneise, 2019-11-29:
[Here in CameraDisplay](https://github.com/cta-observatory/ctapipe/blob/d692d1ed50e3fc597ea9566687d519fc40369359/ctapipe/visualization/mpl_camera.py#L118-L119) the parameter which describes the geometry to be plotted is modified during the plot.

I find this surprising. Is this an actively used feature or can I remove that?

MaxNoe, 2019-11-29:
This should not happen!

MaxNoe, 2019-11-29:
What is setting a mask on a geometry?

kosack, 2019-12-02:
never noticed that! definitely bad behavior. I guess it was added when the pixel masking display was added?  If a Mask is needed, a local version should be stored in CameraDisplay (similar to the `CameraDisplay.image` attribute).

Could we just change it to `self.mask`?   Then it's a mask on the Display, not on the Geometry.

kosack, 2019-12-02:
Actually, I don't really seem the use case for it. I'd suggest just removing it altogether.

kosack, 2019-12-02:
Ah: it's something to do with `GeometryConverter`. I guess that was info that is needed to be recorded between calls, not sure exactly why (need to look).  

https://github.com/cta-observatory/ctapipe/blob/8a1ac929db346c74d8380e2ea765ba7965d6bad5/ctapipe/image/geometry_converter_hex.py#L408

https://github.com/cta-observatory/ctapipe/blob/8a1ac929db346c74d8380e2ea765ba7965d6bad5/ctapipe/image/geometry_converter_hex.py#L524-L543

dneise, 2019-12-03:
Ah I never realized there was such a thing as this geometry converter ... is it in active use? It seems not to be used anywhere in the codebase, only in tests.

Is it supposed to be used directly by ctapipe users outside of Components?? 

If so, should we explain it in a tutorial or an example?

kosack, 2019-12-11:
> Ah I never realized there was such a thing as this geometry converter ... is it in active use? It seems not to be used anywhere in the codebase, only in tests.

It's used by protopipe to be able to apply wavelet filtering, so not used internally, but is a very useful function if you want to apply any image-processing routines that expect square pixels to camera images (e.g. things in scikit-image, or pywavelets, etc)

commit 9cdaedd3 (2020-03-19): Do not mutate geometry in camera display, fixes #1190 (#1237)
closed 2020-03-19 by MaxNoe