cta-observatory/ctapipe #1192: How to plot two CameraDisplays into the same figure with color_bars?

dneise, 2019-11-29:
When I plot a CamDisplay like this:
```python
d1 = CameraDisplay(camera)
d1.image = dl1.image
d1.add_colorbar()
d1.axes.grid()
```

all ist good:
[image]

----

But when I want to plot two next to each other like this:
```python
fig, (a1, a2) = plt.subplots(1, 2, figsize=(12, 6))

d1 = CameraDisplay(camera, ax=a1)
d1.image = dl1.image
d1.add_colorbar()
d1.axes.grid()

d2 = CameraDisplay(camera, ax=a2)
d2.image = dl1.image
d2.add_colorbar()
d2.axes.grid()
```

then both colorbars appear in the 2nd axis
[image]

vuillaut, 2019-11-29:
Hi @dneise 
You must pass the axes to `colobar`:
```
fig, (a1, a2) = plt.subplots(1, 2, figsize=(12, 6))

d1 = CameraDisplay(camera, ax=a1)
d1.image = dl1.image
d1.add_colorbar(ax=a1)
d1.axes.grid()

d2 = CameraDisplay(camera, ax=a2)
d2.image = dl1.image
d2.add_colorbar(ax=a2)
d2.axes.grid()
```
But since `CameraDisplay` should know its axes, maybe that can be improved...

dneise, 2019-11-29:
Ah I did not realize it had that argument ... thx

MaxNoe, 2019-11-29:
Yeah, probably setting ax=self.axes is a saner default in add_colorbar

dneise, 2019-11-30:
Fixed in #1193

dneise, 2019-11-30:
Thanks @vuillaut

closed 2019-11-30 by dneise