cta-observatory/ctapipe #1194: How to write Tool with configurable cleaning?

dneise, 2019-12-01:
Do we already have a configurable Component like `CameraCalibrator` but for cleaning? Something which:

 - exposes the different methods towards the user
 - allows to specify the cleaning thresholds for all possible telescopes
 - allows to specify which method should be used per telescope
 - defines sensible defaults for these things
 - exposes a CLI which is not horrible to use

I would a) need this and b) want to copy it for the Muon Reconstruction Tool I am supposed to write.
This Tool will also have to allow the user to configure a bunch of settings for the different telescopes and define a bunch of useful defaults and have a CLI which is nice to use, i.e. if I want to define some thresholds for all but one telescope, I might want a CLI which can express that.
This will be the same for Cleaning as well as Muon Reconstruction.

MaxNoe, 2019-12-02:
@kosack has it here: https://github.com/cta-observatory/ctapipe/pull/1163

And I think I wanted to open single PRs for these components as they are no all in the dl1 writer itself

kosack, 2019-12-02:
Yes, I'll open PRs about that soon, now that it's stabilized.

dneise, 2019-12-02:
amazing thanks!

dneise, 2019-12-10:
I think this can be closed, #1201 is tackling this problem ...

closed 2019-12-10 by dneise