cta-observatory/ctapipe #1197: Configuration system should be replaced/simplified

kosack, 2019-12-02:
`ctapipe.core.Tool` and `ctapipe.core.Component` are based on `traitlets.config.Configurable` and `traitlets.config.Application` functionality (the config system used by IPython and Jupyter, docs here https://traitlets.readthedocs.io/en/stable/config.html).    It has some nice features, but is pretty horribly implemented and is hard to fix (we patch a lot of its problems in the `Tool` base class).

## Features that are nice:
* configuration options use `Traitlets` that validate themselves and throw meaningful errors if the user mistypes something
* allows configuring from both command-line options and config files
* allows top-level command-line aliases to any low-level config options
* allows sub-commands (but we break that in Tool unfortunately, so currently doesn't work)
* can generate verbose config files from defaults (though undocumented!) 

## Features that could be better
* very poorly documented (many features under-documented or completely un-documented - need to read the code to figure out how they work)
*  config files can only be JSON or python,  would prefer to just support something like YAML (but it's nearly impossible to add a new type without changing the base class) .
* does strange things if no config file is given (#1196 )
* by default doesn't stop in the case of a mistyped config option (default is just to ignore it, which is error-prone)
* really confusing implementation underneath (look at the code here: ) 
* not clear if config file or command-line options have precedence
* only stores the user-specified config on a Class level, not the *full config* that is actually used by instances of classes (that needed to be hacked on by mechanisms like `Tool.add_component()`
* have to set `trait.tag(config=True)` everywhere that should be configurable (would be better to jsut assume all public traits are configurable, but there are a bunch of other ones that get generated by the base class)
* doesn't support positional arguments on the command-line. At the very least, should support notion of *required* parameters, i.e. complain if a parameter with no default value is given. 

## where to start?

I created this Gist to start to explore making a simplified version, but so far it doesn't do everything needed: https://gist.github.com/kosack/8158a2748a458bbece74a010abb02d3c.js

Notably, It's missing the nice chaining of parent classes: config options that are in the parent, should be in a section named by the parent, not the child.

MaxNoe, 2019-12-02:
A bit of googling turned up this: https://github.com/andrasmaroy/pconf

watsonjj, 2019-12-02:
If we do something in-house, would it be worth reconsidering the approach we are using for per-telescope configuration?

A configuration system that provides the configured Components corresponding to the telescope id itself would:
- Allow different algorithms to be used per telescope (e.g. Flashcam's charge extraction) (impossible in current configuration system)
- Keep the internals of the Components simple, as they do not need to aware of different telescopes or subarray layout
- As the pipeline will normally process single-telescope files (at least during the DL1 calibration), its especially unnecessary that the Components require subarray information and handle the per-telescope configuration themselves
- Help remove the telescope loop internal to the CameraCalibrator (something that has been mentioned a few times, most recently https://github.com/cta-observatory/ctapipe/issues/1165#issuecomment-548297793)

This was previously suggested in: https://github.com/cta-observatory/ctapipe/issues/1148#issuecomment-546126553

kosack, 2019-12-02:
> Allow different algorithms to be used per telescope (e.g. Flashcam's charge extraction) (impossible in current configuration system)

For Monte-Carlos this is still an issue. For real data, of course, you run a separate process per file. But yes, I suggest listing any user needs here, so if we do have some time/manpower to make a better one, we could do something that works well.

dneise, 2019-12-03:
>  https://gist.github.com/kosack/8158a2748a458bbece74a010abb02d3c.js

this link seems broken, at least for me on my machine using firefox, you want to remove the `.js` in the end, I think.

adonath, 2019-12-18:
For Gammapy v0.14 we implemented a first version of configuration handling for our high-level `Analysis` class purely based on `jsonschema`. It quickly turned out that this was hard to maintain and extend, because it required editing a complex hierarchical json schema file.

At the last coding sprint we re-visited the question of configuration handling and @cdeil and @Bultako explored available options and wrote minimal prototypes. The result of this study is documented [here]( https://github.com/cdeil/gammapy-config). As a result we agreed to change our configuration system to [pydantic](https://pydantic-docs.helpmanual.io) for now. Here is what the current implementation (for v0.15) looks like: [gammpay config handling](https://github.com/gammapy/gammapy/blob/master/gammapy/analysis/config.py). Right now its only used for config validation, but could probably be extended to be used in the normal Gammapy class API to provide config validation for interactive usage as well.  

In general our impression was the `pydantic` package is nicely written and well documented. 
As far as I can tell pydantic probably does not provide all features required for ctapipe, especially there is no built-in support for command line argument handling. However there are recent feature requests and efforts to get this working:
- https://github.com/mpkocher/pydantic-cli
- https://github.com/samuelcolvin/pydantic/issues/756

I'm not sure `pydantic` is an option worth considering for ctapipe as well (however you might still take a look...). In general I'm interested in which direction ctapipe is going here to keep the possibilities open to share common approaches with gammapy.

kosack, 2020-01-06:
Seems GammPy is now going for *pydantic*: https://pydantic-docs.helpmanual.io/
which does seem nice. Might be worth checking out. 

See also the study they did here: https://github.com/cdeil/gammapy-config

kosack, 2020-01-06:
The nice thing about *Pydantic* is that it makes recursive definitions pretty easy, and not based on the name of the class (as in traitlets.config): 

```py3
from pydantic import BaseModel

class Sub(BaseModel):
    threshold: float

class MyTool(BaseModel):
    output_file : str = "output.txt"
    sub : Sub
   
t = MyTool(sub=Sub(threshold=1.2))

In [63]: t.dict()
Out[63]: {'output_file': 'output.txt', 'sub': {'threshold': 1.2}}    
```

MaxNoe, 2020-01-06:
@kosack yes! 

This makes it so much easier to have things like tools using several instances of e.g. cleaning algorithms

kosack, 2020-01-06:
It would be a bit of work to change everything, but I don't think it would be too complicated.  I have to just check that everything can be made to work on a subset first, but I'm for doing it. Before making such a large change, I will try to write up a proposal first, though.  

It occurs to me that this could also be used in Containers since you can turn off assignment validation (which would be quite slow, and the reason I didn't use traitlets for them originally), and still define our own `Field`-like types to contain the extra metadata for the description and units: we'd only need an `ArrayQuantity` and `Quantity` type probably. 
```

kosack, 2020-01-06:
I have to think about how to support something like `--help`  or `--write-config` in this scheme, however (writing out the whole set of possibilities along with descriptive text without having to construct all classes).  That is a nice feature that would be good to keep. However, since most of the `Component` classes do delayed initialization anyway, it may not be a big problem - we could just construct them all.

MaxNoe, 2020-01-06:
Quick toy example and probably a great idea for a stand-alone add-on for pydantic:

```
from pydantic import BaseModel
import click

def create_cli(model):
    def main(**kwargs):
        model(**kwargs).run()

    for name, option in model.__fields__.items():
        main = click.option('--' + name, type=option.type_, default=option.default)(main)
    main = click.command()(main)

    return main

class Tool(BaseModel):
    name: str = 'World'

    def run(self):
        print('Hello', self.name)

if __name__ == '__main__':
    main = create_cli(Tool)
    main()

```

kosack, 2020-01-06:
Also, it seems that you can include descriptions and other constrants (minimum value for example) like so:

```py3
weight : float = pydantic.Field(0.0, description="this is the weight", gt=0.0)
```

That then ends up in the schema:
```py3
'weight': {'title': 'Weight',
   'description': 'this is the weight',
   'default': 0.0,
   'exclusiveMinimum': 0.0,
   'type': 'number'}},
```

So that can be used to make more descriptive messages in the CLI.   Also, unconstructed classes still have a `.schema()` attribute that works, which means my concern is not valid. So I think it can all work! It does seem a lot nicer than the traitlets scheme.

MaxNoe, 2020-01-06:
Single User code base historically grown vs. an engineered, designed library

kosack, 2020-01-06:
Well, technically *traitlets* is also an engineered library, but I think `traitlets.config` was an afterthought and poorly engineered.

kosack, 2020-01-06:
I did find a big problem that I so far haven't found a solution to: if I want to have other non-configurable variables in a class derived from BaseModel, I cannot.  E.g. let's say I want all Components to have a `x._log` attribute:

The following doesn't work:
```py3
class Component(BaseModel):
   
    class Config:
        validate_assignment=True
    
    def __init__(self, **kwargs):
        super().__init__(**kwargs)
        self._log = logging.getLogger(self.__class__.__name__) # This fails
```
```
ValueError: "Component" object has no field "_log"
```
Adding the following as suggested by the author in PR655 on the pydantic github (wont' link it here to avoid the cross-ref), also doesn't work:
```
    __slots__ = ('_log',) # have to list any non-config vars here, otherwise it doesn't work
```

So that is sort of annoying. These classes seem to be able to *only* have configurable variables, and no other data, since I guess it uses `__dict__` to keep track of them.

MaxNoe, 2020-01-06:
What about

```
@property
def log(self):
    return logging.getLogger(self.__class__.__name__)
```

kosack, 2020-01-06:
That works, but so far in our Component and Tool classes we have many private, non-configurable variables... Might not be so nice to forbid them - it means you can't store any state that doesn't end up in the config file.   For example EventSource needs to store a file handle, etc. 

One option would be to use encapsulation rather than inheritance, but not sure that's very friendly either if you want hierarchical configs.  I have to think about it...  There is a proposed PR to allow an option to ignore private variables beginning with "_", but it doesn't seem to be accepted. That would solve it, probably.

kosack, 2020-01-06:
Perhaps something like what they do with dataclasses: https://github.com/samuelcolvin/pydantic/blob/master/pydantic/dataclasses.py to separate the internal state from the configuration.

MaxNoe, 2020-01-06:
Why not add a `log: logging.Logger` to Component?

kosack, 2020-01-06:
I tried both of those: they work, but then you get the logger in the config output. :

```
enable: true
output_file: out
image_cleaner:
  enable: true
  picture_threshold: 2.0
  boundary_threshold: 5.0
  _log: !!python/object/apply:logging.getLogger
  - TailCutImageCleaner
_log: !!python/object/apply:logging.getLogger
- MyTool
```
And then that breaks writing or reading from JSON, since it's not serializable. 

There is an "exclude" option in the dict() or json() output, but it doesn't seem to work recursively, so if I have a class inside a class, it only excludes the first time that variable appears.

kosack, 2020-01-06:
it's not well documented, but `extra = 'allow'` seems to allow one to create any new variables inside the instance, but they still end up in the config.  With 'ignore', they can't even be created in the instance. Not quite sure what it's supposed to mean.

MaxNoe, 2020-01-06:
Seems about to be solved: https://github.com/sa muelcolvin/pydantic/pull/1139

adonath, 2020-01-07:
Nice @kosack and @MaxNoe that you started looking into Pydantic.

I was not involved much when we made the choice for Gammapy, so I started to learn a bit about Pydantic before the Christmas break as well. When I played around with it I immediately ran into the same issue @kosack described, that instances can not have an internal state, which is not part of the model schema. 

For class attributes there are a few proposed workarounds: `https://github.com/samuelcolvin/pydantic/issues/184`

However the `.dict(include=)` and `.dict(exclude=)` options can be defined recursively, as documented [here](https://pydantic-docs.helpmanual.io/usage/exporting_models/#advanced-include-and-exclude). Which solves part of the problem.

If think what's missing is a way to declare a default selection of variables to serialise. Maybe one could use the `__fields__` attribute to define the defaults and override the `.dict()` along the lines of:

```python3
from pydantic import BaseModel

class MyBaseModel(BaseModel):

    class Config:
        validate_assignment = True
        extra = "allow"
    
    def dict(self, **kwargs):
        kwargs["include"] = self.__fields__.keys()
        return super().dict(**kwargs)

    
class MyModel(MyBaseModel):
    foo : int = 5
    bar : float
        
    def __init__(self, **kwargs):
        super().__init__(**kwargs)
        self.log = "Logger"

m = MyModel(bar=6, bla=17)

print(m.dict())
print(m.log)
print(m.bla)
```
But I'm not sure if this is the way it's supposed to work. I'll include @Bultako in the discussion, as he has the most experience with Pydantic from the Gammapy side.

kosack, 2020-01-07:
> Seems about to be solved: https://github.com/sa muelcolvin/pydantic/pull/1139

That's great! Just yesterday, it looked like it was going to be rejected, but seems things have changed.  I think that would solve the problem!

MaxNoe, 2020-01-07:
I openend another issue, which was closed as duplicate but 4 minutes after that he reconsidered. Coincidence? I think not :rofl:

adonath, 2021-06-04:
I think it might be good to keep the issue alive, so just for the record:
- I think `pydantic` now officially supports private model attributes https://pydantic-docs.helpmanual.io/usage/models/#private-model-attributes (see also `https://github.com/samuelcolvin/ pydantic/pull/1679`)
- Python 3.10 nearly broke `pydantic` (`https://github.com/samuelcolvin/pydantic/issues/2678`), however the Python steering council decided to revert the change that introduced "Postponed Evaluation of Annotations" as a new default in Python 3.10, so that the `pydantic` (and maybe a few other packages relying on type annotations on runtime...) could find a solution in the meantime.

maxnoe, 2025-06-03:
The pydantic situation is now settled to a point where it's very safe to rely on it in my opinion.

There is also now `pydantic-settings` which seems to do most of what we need. I will give implementing our `TelescopeLookup` a go in `pydantic-settings` and if there are no major blockers, I think we should try to switch to pydantic-settings.

I also found the devs very receptive to feature requests: https://github.com/pydantic/pydantic-settings/issues/623