cta-observatory/ctapipe #1198: Use notebook review tool?

MaxNoe, 2019-12-02:
Just stumbled upon this, might be worth using for our documentation notebooks:
https://github.com/marketplace/review-notebook-app

dneise, 2019-12-03:
regarding the notebook-git workflow do you guys have any recommendations? Do you nowadays check in notebooks with content or without? And how do you deal on your local machines with the annoying fact, that notebooks once they were executed locally are shown as modified ... is there any silver lining regarding notebooks and git?

vuillaut, 2019-12-03:
I thought the "rule" was to `nbstripout` the notebooks before to submit so the output is not there but only the modified code appears as modified.
The review notebook app might still be useful :)

kosack, 2019-12-06:
> check in notebooks with content or without

Yes, the idea is you must strip the content - it is then generated when the docs are built by `nbsphinx`, so that we can test for errors.

kosack, 2019-12-06:
Let's try this review tool! I just have to figure out how to install it in the cta-observatory project - When I try via the link, it only gives access to my local repos.

kosack, 2019-12-06:
added

closed 2019-12-06 by kosack