cta-observatory/ctapipe #1200: Transformation of camera coordinates far outside the field of view

MaxNoe, 2019-12-06:
Currently it is possible to transform crazy values of camera coordinates to skycoordinates without a warning.

Then there is a certain point, where it completely breaks down (at pi/2 times the focal length) and the transformation to telescope frame:

```
from astropy.coordinates import AltAz, SkyCoord, EarthLocation
from ctapipe.coordinates import CameraFrame
import astropy.units as u
from astropy.time import Time

altaz = AltAz(
    location=EarthLocation.of_site('Roque de los Muchachos'),
    obstime=Time.now()
)
pointing = SkyCoord(alt=85 * u.deg, az=0 * u.deg, frame=altaz)
camera_frame = CameraFrame(focal_length=5 * u.m, telescope_pointing=pointing)

# crazy, but less than pi / 2
c1 = SkyCoord(x=5 * u.m, y=0 * u.cm, frame=camera_frame)

# more than pi/2
c2 = SkyCoord(x=1.6 * 5 * u.m, y=0 * u.cm, frame=camera_frame)

print(c1.transform_to(altaz))
print(c2.transform_to(altaz))
```

We should probably warn or error if something is further of than something like 15 degrees or so.

kosack, 2019-12-06:
agreed.  Generally this should never happen unless the user is doing something unexpected.    So raising an Exception (perhaps RangeError) is the right way to go, since it is an exceptional case. 

I think adding a line explaining  allowed range to the docstring  or in the high-level user docs would also be useful.  Something like  "The CameraFrame/TelescopeFrame are intended to represent  coordinates within the FOV of the telescope;  coordinates far outside that range may not transform correctly to general Earth or sky coordinates. "

MaxNoe, 2019-12-06:
> agreed. Generally this should never happen unless the user is doing something unexpected. 

It could happen quite regularly for badly reconstructed events. 
Very few events will be that bad but there will be lots of events....

kosack, 2019-12-06:
Ah, yes, i was thinking the reconstructed events are always in the TelescopeFrame, but one might want to plot them in the CameraFrame. In any case, it should still complain.