cta-observatory/ctapipe #1211: consider Astropy 4 and Numpy 0.17

kosack, 2020-01-09:
Astropy 4.0 was just released. It shouldn't cause any major changes to ctapipe, but there is one new feature that is perhaps useful.  If used with NumPy 0.17+, units now work correctly with all numpy ufuncs, which makes some of the hack-work of stripping off units, storing a unit and then re-applying it later in an algorithm no longer needed. 

See http://docs.astropy.org/en/stable/whatsnew/4.0.html#major-improvements-in-compatibility-of-quantity-objects-with-numpy-functions

So in the future, we might consider setting the minimum astropy/numpy version in the package and setup.py so that we can use this.  Currently we have the required versions set lower.

kosack, 2020-01-17:
see #1213 and #1212

kosack, 2020-02-11:
seems to work with both now.

MaxNoe, 2020-02-11:
Actually, I would reopen this issue.

I understood the question not to make ctapipe compatible with the new astropy, but actually *require* it to simplify unit code.

What do you think?

kosack, 2020-02-12:
I'm ok with requiring it! Actually, with the changes we have made, it probably is required now. I haven't tried it with astropy<4.0

closed 2020-05-12 by kosack