cta-observatory/ctapipe #1214: Tools should give a reasonable return value to the shell

kosack, 2020-01-23:
Right now in `ctapipe.core.Tool.run()` different types of errors are caught and the activity status in the Provenance are updated.  However, there are no return values: it should also return 0 on success, and a standard error code if not (See e.g. http://tldp.org/LDP/abs/html/exitcodes.html)

See also: https://github.com/cta-observatory/cta-lstchain/issues/271

kosack, 2020-01-23:
relevant code to modify is here: https://github.com/cta-observatory/ctapipe/blob/12d768d05f6cd279e5b31476da5ae6c9697af437/ctapipe/core/tool.py#L200-L230

MaxNoe, 2020-01-23:
Wasn't this already solved by @dneise? In connections with tool tests failing unnoticed?

I think this is it: https://github.com/cta-observatory/ctapipe/pulls/pull/1027

kosack, 2020-01-23:
Ah, that's good - I think that PR was broken for a while, so got forgotten.  Perhaps we can just get #1027  working and close this then

PR #1027 (2020-04-08): Fix tests for Tools, fixes #1214
Fixes #1025
commit d1010130 (2020-04-08): Fix tests for Tools, fixes #1214 (#1027)

* return some bash return codes, to communicate errors

* assert tool return code is actually zero

* fix tool and thus make test pass again.

* assert return_codes == 0 for all tested tools here

* ignore info() for the moment, this is strange

* add return statements to `tool.run()` calls

* updated to use sys.exit()

* also fix basic test

* Catch tool SystemExit in example notebook

* Adress codacy issues

* Remove return as tools raise SystemExit

* Add test for exit codes

* Use self.exit

* Add run_tool function

* Force pytest color on travis

 [... cut]
closed 2020-04-08 by kosack