cta-observatory/ctapipe #1216: Tool.get_current_config() is broken for TelescopeContainers

kosack, 2020-02-04:
This used to work before the refactoring, so probably isn't too hard to fix.  All TelescopeParameters just return "nan" when queried right now, so you get something like the following when you call get_current_config(), rather than the correct values:

```json
        "TailcutsImageCleaner": {
            "boundary_threshold_pe": null,
            "min_picture_neighbors": null,
            "picture_threshold_pe": null
         },
```

kosack, 2020-02-04:
Seems the problem is that noramally you can get the value of a trait using:

```py3
trait = tool.traits(config=True)['log_format'] . # returns a traitlets.Unicode object
trait.get(tool) # returns the log format string
```
So it just seems that TelescopeParameter is missing that ability, so when get_current_config loops over all configurable traits, it gets a nan value back.

kosack, 2020-02-04:
Bug can be can be reproduced using: 

```py3
def test_works_in_tool(mock_subarray):
    class SomeTool(Tool):
        tel_param1 = FloatTelescopeParameter().tag(config=True)

    tool = SomeTool(subarray=mock_subarray)
    tool.tel_param1=6.0
    config = tool.get_current_config()
    assert config['SomeTool']['tel_param1'] == (["type","*", 6.0])
```

MaxNoe, 2020-02-04:
@kosack The parentheses in your last line are doing nothing, if you want a tuple with a single list there, you'll need a comma after the list.

kosack, 2020-02-04:
Yes - you're right, though that doesn't solve the problem unfortunately! I'll update the example

kosack, 2020-02-04:
I could hack `get_current_config` for the special case of a TelescopeParameter, though there is probably a better way to do it (though again this shows how complex traitlets are, and why we should probably replace the config system)

MaxNoe, 2020-02-04:
This passes:

```
def test_telescope_parameter_tool(mock_subarray):
    from ctapipe.core.tool import Tool

    class SomeTool(Tool):
        tel_param1 = FloatTelescopeParameter().tag(config=True)

    tool = SomeTool(subarray=mock_subarray)
    tool.tel_param1 = 6.0

    config = tool.get_current_config()
    assert list(config['SomeTool']['tel_param1']) == [("type", "*", 6.0), ]
```

So the problem is probably, that the traitlets does not know how to serialize the `TelescopeParameterLookup` object.

MaxNoe, 2020-02-04:
Also: Tool does not have any way to deal with subarrays. Only a `TelescopeComponent` has a subarray argument:

```
ctapipe/core/tests/test_traits.py::test_telescope_parameter_tool
  /home/maxnoe/.local/anaconda3/lib/python3.7/site-packages/traitlets/config/configurable.py:73: DeprecationWarning: Passing unrecoginized arguments to super(SomeTool).__init__(subarray=<MagicMock id='139624136769040'>).
  object.__init__() takes exactly one argument (the instance to initialize)
  This is deprecated in traitlets 4.2.This error will be raised in a future release of traitlets.
    super(Configurable, self).__init__(**kwargs)
```
There is even a typo in the warning

kosack, 2020-02-04:
> So the problem is probably, that the traitlets does not know how to serialize the TelescopeParameterLookup object.

Yes, as mentioned above, it seems to the `Traitlet.get()` function that should turn the traitlet into a "normal" value  (e.g. turn an Integer traitlet back into an int).  Probably there is a simple solution - it's not so well documented though.

kosack, 2020-02-04:
adding a `__repr__()` to `TelescopeComponentLookup` causes it to return a string version that is correct (but it should be a python list of tuples, not a string)

kosack, 2020-02-05:
It occurs to me that perhaps the current implementation of `TelescopeParameter`/`TelescopeParameterLookup` would be better implemented using the `traitlets.observe()` decorator - so TelescopeParameter would just always be a simple list of tuples as now (and it's `klass` would be `list`), but we just register an "observe" function to update/create the `TelescopeParameterLookup` when it changes?   I think that would fix this problem and be cleaner:
See https://traitlets.readthedocs.io/en/stable/api.html#traitlets.observe

PR #1218 (2020-02-11): Fix TelescopeParameter config,  bug #1216
This PR will attempt to fix #1216.   Starting with adding a unit test to check for it, and then a solution.
closed 2020-02-11 by kosack