cta-observatory/ctapipe #1217: master is broken in  test_hyperbinning

kosack, 2020-02-04:
Probably due to a dependency update (numpy or astropy) as before, all master build seem to be failing in the `EventClassifier'  test `test_hyperbinning()`:

```
def test_hyperBinning():
        clf = EventClassifier(cam_id_list=None)
        x = np.array([[26, 70, 53],
                      [97, 20, 56],
                      [35, 38, 81],
                      [48, 60, 40],
                      [73, 68, 63],
                      [96, 86, 63],
                      [73, 67, 6],
                      [48, 66, 60],
                      [47, 82, 87],
                      [60, 52, 74]])
    
        dum_l = [{'maxf': max(x[:, 0]), 'minf': min(x[:, 0]), 'col': 0, 'nbins': 4},
                 {'maxf': max(x[:, 1]), 'minf': min(x[:, 1]), 'col': 1, 'nbins': 2}]
    
        dum_g = clf._hyperBinning(x, dum_l)
    
>       assert np.all(dum_g.size() == (1, 1, 3, 2, 1))
E       assert False
E        +  where False = <function all at 0x7f018c3c2598>(0            ...1\ndtype: int64 == (1, 1, 3, 2, 1)
E        +    where <function all at 0x7f018c3c2598> = np.all
E           -0              1           \n
E           -(26.0, 43.75]  (20.0, 53.0]    1\n
E           -               (53.0, 86.0]    0\n
E           -(43.75, 61.5]  (20.0, 53.0]    1\n
E           -               (53.0, 86.0]    3\n
E           -(61.5, 79.25]  (20.0, 53.0]    0\n
E           -               (53.0, 86.0]    2\n
E           -(79.25, 97.0]  (20.0, 53.0]    0\n
E           -               (53.0, 86.0]    1\n
E           -dtype: int64
E           +(1, 1, 3, 2, 1)
E           Full diff:
E           + (
E           +  1,
E           +  1,
E           +  3,
E           +  2,
E           +  1,
E           + )
E           - 0              1
E           - (26.0, 43.75]  (20.0, 53.0]    1
E           -                (53.0, 86.0]    0
E           - (43.75, 61.5]  (20.0, 53.0]    1
E           -                (53.0, 86.0]    3
E           - (61.5, 79.25]  (20.0, 53.0]    0
E           -                (53.0, 86.0]    2
E           - (79.25, 97.0]  (20.0, 53.0]    0
E           -                (53.0, 86.0]    1
E           - dtype: int64)
```

MaxNoe, 2020-02-04:
What is this hyper binning and for what is this used? 

Where is this EventClassifier actually used?

I would propose the remove this code from ctapipe, until dl1 conversion is merged and a path from dl1.2 to dl2 is explored.

kosack, 2020-02-05:
The EventClassifier just manages the machine learning steps for a set of telescopes for the gamma/hadron separation. 

I think it may be used in Protopipe - @HealthyPear is it still used for the final step of event classification, or was it replaced with a simpler method? I think this hyperbinning was added by  @fvisconti  as well, so perhaps he can comment if it's still useful?  

It's probably quite easy to fix, but if it's not used, we should perhaps think about removing it.

fvisconti, 2020-02-05:
Hi everyone,

sorry for that.

I left the project a couple of years ago (changed job), I can only say it's
maybe not useful anymore and it can be removed.

Cheers!

commit 45bfeb63 (2020-02-06): fix #1217 by removing unused _hyperbinning() function (#1219)

* fix #1217 by removing unused function

the private `_hyperbinning()` function seems to never have  been use anywhere, and since it is broken with Pandas 1.0, it has been removed here.

* also remove level_populations
closed 2020-02-06 by kosack