cta-observatory/ctapipe #1224: Containers, Fields and units

vuillaut, 2020-02-28:
Hi.

I always assumed (but discover this is only at file reading) that the unit in the Containers enforced the setting of compatible units.
I mean that I expected
```
from ctapipe.io import HDF5TableReader, HDF5TableWriter
from ctapipe.io.containers import Container, Field
import astropy.units as u

class MyContainer(Container):
  x = Field(None, unit=u.m)

c = MyContainer()
c.x = 1 * u.TeV
```
to throw an error.

And that I would get
```
c.x = 1 * u.cm
c.x 
Out[1]: <Quantity 0.01 m>
```
Is this something we would like to implement ?

I'd like to have your feedback before trying something.

MaxNoe, 2020-02-28:
Related #445, #501

vuillaut, 2020-02-28:
> Related #445, #501

I should have made a search before asking  🤦‍♂️
Thanks.

kosack, 2020-03-02:
Just to note here, even though it's closed, I think validation on setting a value is probably too costly for performance, but I always intended to implement a .validate() method of Containers that could be called once, e.g. during the schema generation in HDF5TableWriter, just to check that things are ok.  Presumably, since containers are filled for every event, one doesn't need to check every time that something has the right data type and unit. 

In any case, this is not yet implemented, and I've run into problems because it is missing.  E.g. in the current demo of writing the DL1 data, I have the problem that since we use the default value to define the data type, but the default is often np.nan or None, we lose that information, and if the container is not filled correctly on the first event, the schema is incorrect.  So this should be fixed.

closed 2020-10-26 by maxnoe