cta-observatory/ctapipe #1225: add calib_scale as user option in SimTelEventSource

kosack, 2020-03-04:
right now the R1 calibration applied to MC data is done in SimTelEventSource, upon reading.  However, after the refactoring of the calibration and image extraction code, it seems there is no longer an option to specify an overall user-specified multiplicative factor to the calibration anymore.  This. "calib_scale" (in simtel terminology) should become an option in SimTelEventSource, so that it can be configured and included in a config file. 

It should just be included as an overall multiplicative factor during the conversion from DC to PE.  Usually this factor includes just the difference between "mean PE" units and "peak PE" units (the former being what the MCs use, and the latter is what  is measured in the lab).

I guess the default can just be 1.0 (no effect).

MaxNoe, 2020-03-04:
Wouldn't this be solved by calculating the conversion coefficients on simulated data the same way as on observed data instead of relying on some other coefficients from simtel_array?

This anyhow needs to be done, if extraction algorithms, window size and so on differ between the simtel and the ctapipe implementation, which since we have several algorithms with lots of parameters will more often be the case then not?

kosack, 2020-03-04:
Yes, but that means changing the way the simulations are configured, so probably must be left to a later improvement. Right now we have no simulated DC/PE flasher data that could be used for that.  

In any case, adding a calib_scale option lets you add any fudge factor you want, not just this conversion, so it is still useful, even for things like testing the effects of miscalibration.   I agree it would be better to just do things the same way for both in the analysis part, but it still probably has to appear somewhere, since the simulation needs it in peak units as an input (so it would have to do the conversion).

MaxNoe, 2020-03-04:
>  Right now we have no simulated DC/PE flasher data that could be used for that.

Ah, I thought this would be the calibration events I implemented some time ago in eventio.

#calibration https://github.com/cta-observatory/pyeventio/pull/152

FrancaCassol, 2020-03-05:
> I agree it would be better to just do things the same way for both in the analysis part, but it still probably has to appear somewhere

Hi @MaxNoe, @kosack 
actually it is appearing already for LST1 data (see my slides of last lST1 analysis meeting [https://indico.cta-observatory.org/event/2664/](https://indico.cta-observatory.org/event/2664/contributions/23185/attachments/17084/22547/FF_MC_simulation_FCassol_1_3_2020.pdf) ). This is related to PR [https://github.com/cta-observatory/ctapipe/pull/1115](https://github.com/cta-observatory/ctapipe/pull/1115) (actually, what is blocking it?) I am at present using the LightEmission package but I will have a look on how to use the simulated laser events that are now accessible thanks to [https://github.com/cta-observatory/pyeventio/pull/152](https://github.com/cta-observatory/pyeventio/pull/152) .

watsonjj, 2020-03-07:
https://github.com/cta-observatory/ctapipe/issues/676#issuecomment-375468358

> Integration correction and calib_scale are two different correction factors. The former corrects for the part of the pulse shape thats not within the integration window, the latter corrects for the different definitions of what "1 photoelectron" is.
> 
> In the lab we calibrate to photoelectrons by looking at the spectrum with low illumination, so we see the single pe peaks, and use the x value of the 1pe peak in that spectrum. Instead in the MC (I think this is true) pe is defined from the mean of the SPE spectrum input into simtelrray. So to convert between the two you somehow need access to the SPE spectrum that was input to create the simulation and/or create the SPE spectrum from the low pe waveforms of the MC so that you calibrate them in a similar way lab data is calibrated.
> 
> Another technique is by looking at the bias in the charge resolution, and removing it by altering calib_scale, but in my opinion that suggests there are no other factors that could be creating the bias, I think the SPE spectrum method is more correct, and is typically how one (or atleast how we, in GCT) calibrates lab data into photoelectrons.

calib_scale exists due to inconsistencies in the definition of 1 photoelectron. Some (including simtelarray) define it as the average of the charge spectrum when you have for certain only 1 p.e. (i.e. the input spectrum for simtelarray before it is normalised). Others (e.g. HESS) define it as the position of the first p.e. peak (maximum bin) in the SPE spectrum.  calib_scale changes from the former to the latter, and is only needed when cameras calibrate their real data using the latter definition of p.e..

CTA should enforce that all cameras use the former p.e. definition. Then p.e. is standardised and is common with simtelarray (and removes the need for calib_scale). It is also the only photoelectron definition that makes sense for SiPMs, due to optical crosstalk.

MaxNoe, 2020-05-07:
Was this resolved by removing calib scale entirely? I cannot find a single occurance of it in the current master.

@watsonjj

watsonjj, 2020-05-07:
calib_scale was removed a while before this issue was created. As per my explanation, I believe it to an unnecessary complication in CTA. I was waiting for further discussion on the matter before closing this.

MaxNoe, 2020-05-07:
Okay, then let's close this. 

If anyone has any objections, feel free to reopen

maxnoe, 2026-10-12:
This was, despite @watsonjj conclusion here, added to ctapipe for comparison with CTAMars and EventDisplay:

- https://github.com/cta-observatory/ctapipe/pull/1749

LST uses a channel dependent calib_scale (calib_scale_high_gain, calib_scale_low_gain) to correct on an average basis, for how their integration window interacts with the pulse shape (similar to the `integration_window_correction`).

To be comparable to the LST analysis, we would have to move the `calib_scale` to the `CameraCalibrator` instead of the `SimTelEventSource` and add seperate options per gain channel.

closed 2026-11-02 by maxnoe