cta-observatory/ctapipe #1226: Proposal: Refactor FlatFieldCalculator and PedestalCalculator

watsonjj, 2020-03-07:
The `FlatFieldCalculator` and the `PedestalCalculator` that are currently located in the `calib` submodule are not directly used for calibration. Instead they are used for monitoring and analysis of the charge statistics. From these values it is possible to extract the calibration coefficients, but this is secondary.

Additionally, these classes are fairly large and complicated as they instance their own `ImageExtractor`, something I think should be discouraged. To ensure a common utilisation of `ImageExtractor` and other calibration, I would suggest that a correctly configured `CameraCalibrator` should be provided to such classes, as I describe in #1147.

As part of #1026, and in order to achieve the second pass described in https://github.com/cta-observatory/ctapipe/issues/1147#issuecomment-546264078 I intended to implement the calculators which directly calculate and fill the calibration coefficients created in #1160. However, I am finding that these current "Calculator" classes are obstructing and would cause confusion with the calibration coefficient calculators I intend to contribute.

I propose that the calculators that exist in the `calib` submodule should only directly calculate the calibration coefficients. The classes should also not instance their own `ImageExtractor`. The classes as they currently exist, to monitor the charge/pedestal statistics, should be moved elsewhere.

kosack, 2020-03-09:
Are you talking about on-the-fly computation here of pedestals and FF? Or just to ensure that the current off-line FF and Pedestal calculators use the same CameraCalibrator internally?

watsonjj, 2020-03-09:
My main concern is the `FlatFieldCalculator` and `PedestalCalculator` don't calculate calibration coefficients. Their purpose appears to be more towards "charge statistics monitoring". Further analysis or "calculation" has to be done on the return values to extract the calibration coefficients.

FrancaCassol, 2020-03-09:
Hi @watsonjj ,
these modules are the base Components to fill the monitoring containers used to calculate the calibration coefficients (which are then finalised in a tool based on them). They are presently in use and represent the core components for the LST1 calibration (which works successfully). Are you really sure you can not propose a development without touching them? At least till you do not assure an equivalent calibration level ;-)

watsonjj, 2020-03-09:
This is a problem I was conscious of, and is why I am scared of touching these classes. I don't think progress in development of ctapipe should be stalled because it would break specific cameras pipelines. With this approach ctapipe will never be able to progress. It is clear from the open issues that there are lots of areas that may change about ctapipe, which will break current api.

It should be made clear that external software should only depend on releases of ctapipe, not the master. And if the master is relied on, there is no guarantee things won't change which would break external code....

rlopezcoto, 2020-03-09:
Hi, just to clarify: in the LST1 software we do as @watsonjj suggests: [lstchain](https://github.com/cta-observatory/cta-lstchain) master depends on ctapipe releases. I guess what @FrancaCassol is afraid is that these changes break some of these modules functionalities even for future ctapipe releases.

watsonjj, 2020-03-09:
I will start by contributing a new calibration class, without touching these Calculator class. Maybe then we can work to converge the calibration classes to have a fully implemented calibration in ctapipe (and not spilt between here and lstchain)