cta-observatory/ctapipe #1243: Extracting pulse width

watsonjj, 2020-03-24:
The pulse width is potentially a useful parameter to extract from the waveform. A longer pulse width may suggest a time-extended Cherenkov emission, potentially from a shower with heavy progenitors (e.g. iron showers). We may therefore wish to include the extraction of the pulse width as part of the ImageExtractor (much in the same way as charge and pulse time is extracted), and include it in the DL1 container.

The difficulty with this exercise is to do this in an efficient way, so as to not heavily impact the runtime of the ImageExtractor algorithms. This is likely to need the use of the numba package.

kosack, 2020-03-24:
Yes, this is a good idea. The pulse can be fully defined by peak, width, risetime, and falltime, so it could be useful to compute all of them here (we did this in VERITAS, at least for testing).

note that the width can be defined in many ways as well - is it the width at 50% of the peak?  A few time measurements are useful:

* t_thresh: time the pulse first crosses a user-defined threshold for peak-detection (useful to measuring time jitter)
* t_50 [leading, trailing]: time position at 50% of the pulse on the leading-edge or trailing edge 
* t_90 [leading, trailing]: time position at 90% of pulse for leading and trailing edges
* t_10 [leading, trailing]: time position at 10% of pulse for leading and trailing edges

Then
* width = t_50[1] - t_50[0]. (1=trailing edge, 0=leading edge)
* rise_time = t_90[0] - t_10[0]
* fall_time = t_10[1] - t_90[1]

kosack, 2020-03-24:
so the main function needed is one that finds the time in the waveform where the value of the waveform crosses a threshold, either forward (from start of search window) or backward (from end of search window). That allows you to calculate all of these times. 

Something like:

```py3
def find_waveform_time(waveform, threshold, forward=True):
   ...

t_50_0 = find_waveform_time(waveform, peak/2.0, forward=True)
t_50_1 = find_waveform_time(waveform, peak/2.0, forward=False)
```
where the return values are the time for each pixel in the image.

watsonjj, 2020-03-24:
I had a much-less-than-perfect algorithm to extract the FWHM and rise time of the pulse implemented a while ago in CHECLabPy. It uses numba, but it still impacted the event processing speed enough that I avoided using it.

https://github.com/sstcam/CHECLabPy/blob/master/CHECLabPy/waveform_reducers/timing.py

It could be a useful resource though.

kosack, 2020-03-24:
The simplest method is just this, but not sure how efficient it is:

```py3
@njit  # to avoid intermediate value
def find_waveform_time(waveform, threshold):
    """ returns sample number of first time waveform crosses threshold """
    return np.argmax(waveform > threshold, axis=1)
```

MaxNoe, 2020-03-24:
Is numba really make a single call to a numpy method faster?

watsonjj, 2020-03-24:
That would also encounter problems for thresholds close to the baseline - you would get the first sample that meets the condition, not necessarily the one closest to the pulse (which is more likely to be correct)

kosack, 2020-03-24:
It works fine as long as the threshold is not too close to the noise.  For the width, it's ok - if the pulse is in the noise, then width will be crazy values, but you can filter them out.  You could of course pass only the part of the waveform near the pulse, if you have a good estimate of that window. 

> Is numba really make a single call to a numpy method faster?

probably not needed, but it avoids a temporary being created for the X>Y operation.

MaxNoe, 2020-03-24:
> probably not needed, but it avoids a temporary being created for the X>Y operation.

ah, that's nice

kosack, 2020-03-24:
> That would also encounter problems for thresholds close to the noise - you would get the first sample > that meets the condition, not necessarily the one closest to the pulse (which is more likely to be correct)

I think most of the time, you'd only look at width, rise_time, and fall_time values for pixels that have a strong signal or pass cleaning later, so I think it's ok to not be perfect here. Denoising comes later.

watsonjj, 2020-03-24:
> The only thing is that this is only sample-accurate. If you want better accuracy, you would need to use a resampled image or do a spline, which probably would be quite a bit slower.

A simple interpolation could also be sufficient, right?

kosack, 2020-03-24:
I guess you could do that with numpy as well, once you have the sample-accurate version, just do a linear interpolation between sample and sample-1. Not sure it makes much of a difference, though, so might not be needed.