cta-observatory/ctapipe #1247: Improve Bokeh Camera Display

kosack, 2020-04-01:
There is a Bokeh version of the CameraDisplay in `ctapipe.visualization.bokeh.CameraDisplay` now that is much faster than the default MatPlotLib one, and is better suited for interactive use (the matplotlib version is better for making nice PDFs, etc).  However, it's currently not used very often since it's not friendly to use in a notebook, and doesn't have as nice a display as the default one. 

Here I have created an example notebook that demonstrates that we can do it as nicely as the MatPlotLib version, and have it work in a Jupyter Notebook (also in Jupyter Lab).

https://gist.github.com/kosack/aceb8b6a8c6ab616767759ca48609063

Download this notebook and run it to try it out (it doesn't work via GitHub's notebook renderer)

The resulting display looks like this,  handles both hexagonal and square pixels, and updates at > 30 frames per second on my laptop, making it really nice for interactive functions using `ipywidgets.interact()`:

[image]

Zooming works properly and in real-time:
[image]

Right now the code is just a proof-of-concept snippet, but I think we could use it to improve the current implementation and automatically push updates to a Notebook for realtime use.  That might break the functionality of the current non-notebook version, which so far is only used in `ctapipe-event-viewer`, which could be modified to be a fully in-notebook display, not a separate application. Or we can make it work in both cases. 

A further development would be to make ctapipe.visualization.CameraDisplay be just an  dummy class that instantiates either an `mpl.CameraDisplay` or `Bokeh.CameraDisplay` upon request (either by delegation, or subclassing).

MaxNoe, 2020-04-01:
At this point we could then also think about making these two libraries optional dependencies (matplotlib and bokeh). At least one of them should be optional as both are pretty hefty (but matplotlib is more widely used at the moment.)

kosack, 2020-09-14:
incomplete, but perhaps more useful example

```py3
import numpy as np
import bokeh
from bokeh.io import output_notebook, push_notebook, show
from bokeh.plotting import figure
from ctapipe.instrument import CameraGeometry

def _generate_polygon_vertices(order=6, x=0, y=0, size=1.0, theta0=0.0):
    """
    returns the vertices of a regular polygon of order `order`,
    centered at `(x,y)`, with size `size` and rotation angle `theta0` in deg.
    """
    angles = (np.linspace(0.0, 360.0, order + 1) + theta0) * np.pi / 180.0
    return size * np.cos(angles) + x, size * np.sin(angles) + y

def _generate_bokeh_multi_polygon_arrays(geom, order=6):
    pixel_polys_x = []
    pixel_polys_y = []
    for pix_x, pix_y, pix_area in zip(
        geom.pix_x.value, geom.pix_y.value, geom.pix_area.value
    ):
        vx, vy = _generate_polygon_vertices(
            order=order,
            x=pix_x,
            y=pix_y,
            size=np.sqrt(pix_area * 2) * 3 ** (-3 / 4),  # only for hexagon
            theta0=geom.pix_rotation.deg + 180 / order,
        )
        pixel_polys_x.append([[vx]])
        pixel_polys_y.append([[vy]])
    return pixel_polys_x, pixel_polys_y

class CameraDisplay:
    """
    CameraDisplay implementation in Bokeh
    """

    def __init__(self, geom: CameraGeometry, use_notebook=True, autoshow=True):

        self._geom = geom
        self._use_notebook = use_notebook
        self._handle = None

        order = 6 if geom.pix_type == "hexagonal" else 4
        xs, ys = _generate_bokeh_multi_polygon_arrays(geom, order=order)

        self.datasource = bokeh.plotting.ColumnDataSource(
            data=dict(
                poly_xs=xs,
                poly_ys=ys,
                image=np.ones_like(geom.pix_x.value).astype(np.float),
            )
        )

        self._color_mapper = bokeh.models.mappers.LinearColorMapper(
            palette=bokeh.palettes.Viridis256, low=-10, high=40
        )

        self.figure = figure(
            title=f"{geom} ({geom.frame.__class__.__name__})",
            match_aspect=True,
            aspect_scale=1,
        )
        # nmodifies the BoxZoomTool. May be better way to get it than by index
        self.figure.toolbar.tools[2].match_aspect = True
        self._setup_camera(geom)

        if use_notebook:
            output_notebook()

        if autoshow:
            self._handle = show(self.figure, notebook_handle=use_notebook)

    def _setup_camera(self, geom):
        self._pixels = self.figure.multi_polygons(
            xs="poly_xs",
            ys="poly_ys",
            fill_color=bokeh.transform.transform("image", self._color_mapper),
            line_color=None,
            source=self.datasource,
        )
        self._color_bar = bokeh.models.ColorBar(
            color_mapper=self._color_mapper,
            label_standoff=12,
            border_line_color=None,
            location=(0, 0),
        )
        self.figure.add_layout(self._color_bar, "right")

    def update(self):
        if self._use_notebook and self._handle:
            push_notebook(self._handle)

    def rescale(self, percent=100):
        self._color_mapper.update(
            low=self.datasource.data["image"].min(),
            high=(percent / 100) * self.datasource.data["image"].max(),
        )

    def set_geom_and_image(self, geom, image):
        self.geom = geom
        self.image = image

    @property
    def geom(self):
        return self._geom

    @geom.setter
    def geom(self, new_geom):
        self._geom = new_geom

    @property
    def image(self,):
        return self.datasource.data["image"]

    @image.setter
    def image(self, new_image):
        self.datasource.data["image"] = new_image
        self.update()
```

MaxNoe, 2020-09-14:
The vertices can even be broadcasted to avoid those loops;.

moralejo, 2020-09-15:
Probably easy, but I do not manage to do it: how could one make the high-low ends of the color scale change when bin contents change? I tried adding those to the datasource, but one cannot add single values, only arrays of length = # of pixels.

MaxNoe, 2020-09-15:
I don't know how to do it in JS in the finished html, but in python you need to update the `low` and `high` attributes of the color mapper like so:

```
# initial code
cmap = LinearColorMapper('Inferno256', low=image.min(), high=image.max())

# update to next event:

def update_image(new_img):
    # replace image data in datasource with new image
    source.patch({'image': [(slice(None), new_img)]})
   
    # update color mapper
    cmap.low = new_img.min()
    cmap.high = new_img.max()
```

maxnoe, 2022-09-23:
Also implemented in #1494

closed 2022-09-23 by maxnoe