cta-observatory/ctapipe #1252: Missing properties of telescope optics for muon analysis

MaxNoe, 2020-04-04:
The muon analysis has currently hard coded values for several properties of the telescopes that or not in the optics description.

* camera radius
* central hole radius
* primary mirror diameter
* secondary mirror diameter

moralejo, 2022-12-14:
I *think* we should use the coordinates of the center of the mirror for the stereoscopic reconstruction. In Mars I believe we assume the center of the mirror and the center of rotation are the same point, so we always use the same coordinates regardless of pointing. But in reality they are like 0.5 m away from each other. I don't know how big the error introduced by that assumption is.

maxnoe, 2026-11-10:
Most of these are now covered by the detailed optics description introduced in #3085.

I close here and open a couple of follow ups.

closed 2026-11-10 by maxnoe