cta-observatory/ctapipe #1257: Hard-coded values in muon ring intensity likelihood maximization

moralejo, 2020-04-07:
https://github.com/cta-observatory/ctapipe/blob/4316ae183c0eaa88d40f3259651be4f3efad9df3/ctapipe/image/muon/muon_integrator.py#L337

The values of the 3rd and 4th parameters, spe_width and ped (pedestal RMS) should be configurable. With proper values, the chi2 of the muon intensity fit will be meaningful, and can be used for a sort of "excess-noise factor test" performed with the muon rings.

MaxNoe, 2020-04-07:
I'm currently working on making this a proper configurable component,  should be done today or tomorrow

moralejo, 2020-04-07:
> I'm currently working on making this a proper configurable component, should be done today or tomorrow

Ok, great. Note that while spe_width (related to the F-factor) is indeed a value to be passed through a config file, the other should be read from the calibration output: it is the rms of the pedestal for the unbiased (GlobalPeakWindowSum) extractor used in the muon ring analysis. The calibration uses the same rms for the estimate of conversion factors via the  F-factor method.

moralejo, 2020-04-07:
> Ok, great. Note that while spe_width (related to the F-factor) is indeed a value to be passed through a config file, the other should be read from the calibration output: it is the rms of the [...]

Also: the pedestal one should be an array, since large pixel-wise differences are expected due to stars.

MaxNoe, 2020-04-07:
This should just be read from the monitoring data, right?

kosack, 2020-04-07:
> This should just be read from the monitoring data, right?

so far that is not defined, though, so might be best to make it a configuration parameter at first and later we can refactor it to automatically load it from monitoring if available.

moralejo, 2020-04-07:
> This should just be read from the monitoring data, right?

The unbiased pedestal RMS should already be available in the monitoring data from the calibration stage. Perhaps @FrancaCassol can comment. Eventually one should also make sure it is for the same window_width as used in the pulse extraction for muon rings (if it is not, anyway it will be better than the current fixed value, at least will account for differences among pixels).

kosack, 2020-04-08:
> The unbiased pedestal RMS should already be available in the monitoring data from the calibration stage. Perhaps @FrancaCassol can comment. Eventually one should also make sure it is for the same [...]

By the way, that means that we must write monitoring files with enough header info that we can tell what settings were used.  At least including the user config in the file would be useful (which is already part of the provenance info).  That's a general comment for the DL1 files as well.

MaxNoe, 2020-04-08:
I added this as a second image shaped or single number parameter to the fitting function here #1261 

Where this value comes from, config or monitoring data can then be chosen by the application

MaxNoe, 2020-04-28:
Fixed in #1261

closed 2020-04-28 by MaxNoe