cta-observatory/ctapipe #1258: Ring completeness is quite wrong

MaxNoe, 2020-04-07:
[image]

Might just be the assumption of a circular camera. Should be replaced.

rlopezcoto, 2020-04-07:
it depends on the threshold that you put per "sector". I think that the default is 30 phe per each of the sectors in which the ring is divided

moralejo, 2020-04-07:
> it depends on the threshold that you put per "sector". I think that the default is 30 phe per each of the sectors in which the ring is divided

Indeed. There is some certain arbitrariness in the definition. Once we have the actual pedestal rms we might do it in terms of significant excess above NSB, rather than above a fixed threshold in p.e., so it will be more in agreement with "visual judgement".

On the other hand this ring completeness ia rather unnecessary in the end. "containment" is more relevant, since an "incomplete" ring which is _not_ clipped by the FoV can be pretty well reconstructed. In contrast, for a 0.75-complete ring in which the missing 0.25 is outside the camera we cannot reconstruct the impact parameter reliably.