cta-observatory/ctapipe #1260: Misleading names of coordinates in Telescope and Nominal Frame

MaxNoe, 2020-04-07:
@moralejo made a good point that `delta_alt` and `delta_az` imply that `delta_alt = altaz1.alt - altaz2.alt`.

Which is not the cases, as these are in a rotated coordinate frame and this holds only for the center point.

We should come up with more appropriate names or just go back to x and y and then make sure that x and y are aligned with the axes in the CameraFrame.

Opinions?

kosack, 2020-04-07:
I don't like x and y, since this is not a cartesian frame. The should be at something that show that  they are angles on a sphere. 

I don't really understand the problem with with delta_alt and delta_az, since the delta is relative to the center of the camera (or pointing in the NominalFrame).  Doesn't that just imply a better docstring?  They are not rotated with respect to the center points in alt/az (except for pointing corrections).  If the pointing position is (az,alt) = (220d, 40d), these are coordinates that are relative to that point.

moralejo, 2020-04-07:
> I don't like x and y, since this is not a cartesian frame. The should be at something that show that they are angles on a sphere.
> 
> I don't really understand the problem with with delta_alt and delta_az, since the delta is relative to the center of the camera (or pointing in the NominalFrame). 

My only problem is "az" may be confused with the azimuth of the horizontal coordinates. This is instead an "azimuth" in a system in which altitude=0 corresponds to the current FoV center.

> Doesn't that just imply a better docstring? They are not rotated with respect to the center points in alt/az (except for pointing corrections). If the pointing position is (az,alt) = (220d, 40d), these are coordinates that are relative to that point.
> 
> A TelescopeFrame point with delta_alt=4.0d is really 4 deg above the center point.

yes, and a point with delta_az=4 deg is 4 deg away from central point regardless of the altitude of the FoV center.  So it is very close to the x-y system in the camera, in angular units.

I have no problem with delta_alt because the altitude of (alt,az) is measured along a great circle, so delta_alt is not misleading. But delta_az is a bit misleading. Pointing at zenith the camera points at delta_az = -1,  +1 deg would have ("standard") azimuth values differing by 180 deg.
This said, if you prefer not to use delta_x, delta_y, I would indeed be happy with a clear docstring.

kosack, 2020-04-07:
Yes, I agree delta_az is actually wrong (since there is the cos(alt) factor to deal with).  I think we had this discussion when making the system, and didn't want to use theta/phi since they were already over-used elsewhere. Perhaps there are better standard symbols for this sort of frame defined somewhere. I dislike X,Y, since that implies a projection, and these are points on a sphere with no projection.

kosack, 2020-04-07:
This can't be an uncommon frame to use... I wonder if there is some naming standard elsewhere? Unfortunately I could only find projected coordinates (like for a CCD camera), where it is "DETX, DETY".

MaxNoe, 2020-04-07:
We could just go with generic lat/lon this would make it clear that it's a spherical frame and don't imply any wrong assumptions about relations to other frames, be it to horizontal by choosing alt az or camera by choosing x y

kosack, 2020-04-08:
> We could just go with generic lat/lon this would make it clear that it's a spherical frame and don't imply any wrong assumptions about relations to other frames, be it to horizontal by choosing alt az or camera by choosing x y

Except lat/lon has the same problem: degrees in lon must be scaled by cos(lat) to get physical degrees on the sky sphere.

MaxNoe, 2020-04-08:
That was not the problem @moralejo was talking about. This holds true for every spherical coordinate frame, so this cannot be a problem.

The problem @moralejo was having is that the name implied a relation to the alt az in the horizon system that did not hold true.

kosack, 2020-04-08:
> The problem @moralejo was having is that the name implied a relation to the alt az in the horizon system that did not hold true.

Ok, It's just az that is not correct, so lat/lon isn't too bad as long as we document that it is (lon,lat) where the pointing (az,alt) is (0,0) position and it is rotationally aligned with the HorizonFrame

kosack, 2020-04-08:
Just to note, the original choice if delta_alt and delta_az was really meaning a lat/lon from an alt/az position. But it wasn't considered that it could be misinterpreted

MaxNoe, 2020-04-08:
> aligned with the HorizonFrame.

Actually, it's not really aligned with the horizon frame, only at alt=0. We should care about a fixed / easy to understand alignment with the camera frame. so lat parallel to x and lon parallel to y for example.

kosack, 2020-04-08:
I'm ok with changing to  lat/lon and updating the docs to be clear where the reference position is (and agreed that referring to the pointing as a representation in alt/az is bad), or the more wordy `lat_from_pointing`, and `lon_from_pointing`

MaxNoe, 2020-04-14:
@kosack the open gamma astro formats also use lat/lon as in `FOV_ALTAZ_LATLON`

https://gamma-astro-data-formats.readthedocs.io/en/latest/general/coordinates.html

commit 71ebb473 (2020-04-28): Rename coordinates of NominalFrame and TelescopeFrame, fixes #1260 (#1299)

* Rename coordinates of NominalFrame and TelescopeFrame, fixes #1260

* Add copy=False in coordinate trafo
closed 2020-04-28 by MaxNoe