cta-observatory/ctapipe #1263: Quantity telescope component

MaxNoe, 2020-04-07:
There is currently no support for a quantity telescope component.

I needed that for integration window in meters

kosack, 2020-04-07:
Do you mean TelescopeParameter?

MaxNoe, 2020-04-07:
yes, sorry

kosack, 2020-04-07:
You want to be able to set the value by a quantity string, I guess? You could always just name the parameter "window_m" to be clear it's in meters.  Or for very basic functionality, you could do:

```py3
window_width = TelescopeParameter(dtype=u.Quantity...)
```
but that will not check that it's the right type of unit.

MaxNoe, 2020-04-07:
`lambda_min = TelescopeParameter(dtype=lambda x: u.Quantity(x, u.nm)`

could work

kosack, 2020-04-07:
Not sure that would survive the round-trip to a config file and back though.  It's somewhat difficult to do with objects that don't have a JSON or YAML representation

MaxNoe, 2020-04-07:
Json for a quantity would be a dict of value and unit which we could easily handle in a json encoder / decoder

gabemery, 2022-07-28:
Is anything planned related to this issue or support of Quantity is not expected?

maxnoe, 2022-07-28:
I don't know of any current efforts to close this issue