cta-observatory/ctapipe #1270: What is telescope type id?

MaxNoe, 2020-04-10:
```
ctapipe/containers.py
76:    tel_type_id = Field(0, "telescope type id number (integer)")
```

Why is there a telescope type id defined in the container for telescope_index ? What is this number supposed to mean and where to fill it?

kosack, 2020-04-10:
This is used in the dl1 writer.  Right now it's jsut an integer hash of the telescope name, since there is no numbering or naming scheme yet.  It's only for being able to do things like:

```py3
parameters_dataframe.groupby("telescope_type")
```

In the case of merged tables (where all events are in one table)

kosack, 2020-04-10:
There is some confusion, since I've been slowly breaking #1066  into smaller PRs, and that one was merged a while ago, but is only used in #1066 so far.

kosack, 2020-04-10:
If we can fix the telescope naming scheme, we could make that a string or Enum later. But that also requires adding support for strings in HDF5TableWriter (which also means adding a few attributes to Field to be able to set the storage length, etc).

kosack, 2020-04-10:
```py3
def tel_type_string_to_int(tel_type):
    """
    convert a telescope type string (str(TelescopeDescription)) into an integer that
    can be stored.

    Parameters
    ----------
    tel_type: str
        telescope type string like "SST_ASTRI_CHEC"

    Returns
    -------
    int:
        hash value
    """
    return np.int32(
        int(hashlib.sha1(tel_type.encode("utf8")).hexdigest(), 16) % (10 ** 8)
    )
```

MaxNoe, 2020-04-10:
why not just `hash(str(tel_type))`?

kosack, 2020-04-10:
> why not just hash(str(tel_type))? 

it was just to get a shorter more readable number, and to make sure it comes out to the same number on all machines. I'm not sure hash() guarantees that.

MaxNoe, 2020-04-10:
Is that function already in the current master?

kosack, 2020-04-10:
No, just in #1163 but I'm working to make some PRs to move those in.   Now that we split the DL1 and other files by tel_id, even for things like parameters, that `tel_type_id` may not be needed anymore.   In the original DL1 format, I had all hillas and other parameters in the same table, since they are telescope independent, but we decided against that.

MaxNoe, 2020-05-18:
Tel type id was removed

closed 2020-05-18 by MaxNoe