cta-observatory/ctapipe #1274: Keras for shower reconstruction

capaths, 2020-04-13:
Hello, I'm from a team working on a shower reconstruction technique based on neural networks using Keras. We think it'd be nice to be able to use Keras models to create a shower reconstructor.

Currently we have a model that uses as input the calibrated data of an event from prod3b  using a custom [Sequence](https://www.tensorflow.org/api_docs/python/tf/keras/utils/Sequence) ([here](https://github.com/sborquez/gerumo/blob/master/gerumo/data/generator.py)) that allows the neural network to have a custom input and output. Maybe using a Sequence inside a new Reconstructor sub-class could help to have a common input and output in the `predict` method of that class for different reconstructors based on Keras.

MaxNoe, 2020-05-19:
Related: #1091

I would be reluctant to add one of the deep learning libraries as a dependency.
But as far as I know, all of them support exporting to onnx (at least tf and torch).

So what would you think about using onnx as serialization format?

capaths, 2020-06-16:
Sorry for the late reply. I've been testing the conversion of our models used for shower reconstruction to the onnx format using [keras2onnx](https://github.com/onnx/keras-onnx) and it seems to be working fine, but I had to make some adjustments to the code (nothing that changes the model architecture though).

ONNX as the serialization format seems like a good idea, as it could allow models from different frameworks, not only Keras, to be loaded and, compared to Tensorflow for example, doesn't seem too heavy to have as a dependency.

A while ago, we started to develop, as a contribution to ctapipe, a "Keras Reconstructor" for the usage of one or more models in a shower reconstruction. We'll be using this reconstructor as a base to an "ONNX Reconstructor" (or maybe "Deep Learning Reconstructor"?) and create a pull request when it's ready (in a couple of days).

closed 2023-01-04 by maxnoe