cta-observatory/ctapipe #1275: use a robust fitting in timing_parameters

kosack, 2020-04-13:
Right now the computation, using `np.polyfit()` of the timing axis in timing_parameters can be quite biased if there are outliers included (some noise included after cleaning, which happens relatively often).  It was suggested here  https://github.com/cta-observatory/ctapipe/issues/602#issuecomment-350823139 that we should use a robust fitting routine (e.g. one that is not strongly affected by outliers), and this is already implenmented in MARS and EventDisplay. 

Here is a comparison of the results of fitting a linear function  with polyfit to several robust methods included in SciPy:

```py3
    N = 15
    x = np.arange(N)
    y = 3*x+2 + np.random.uniform(5.0,size=N)-2.5
    
    # add some outliers
    y[outlier_pos] = 3*x[4]+2 + outlier_level
    
    pslope, pintercept = np.polyfit(x,y,1)
    tslope, tintercept, err_slope_low, err_slope_high= stats.theilslopes(y,x=x)
    sslope, sintercept, = stats.siegelslopes(y,x=x, method='hierarchical')
    ssslope, ssintercept, = stats.siegelslopes(y,x=x, method='separate')
```

[image]

I suggest we replace `polyfit`  with `scipy.stats.siegelslopes` to get a better fit.  The only problem is that these functions do not return error estimates, so those must be computed separately. 

This is also mentioned in #1215

MaxNoe, 2020-04-13:
>  The only problem is that these functions do not return error estimates, so those must be computed separately.

In think we could probably also live without these here. Or just a simple rmse (which again would be affected by outliers).

kosack, 2020-04-14:
yes, so far I don't think we use the error estimates, so could just drop them.

MaxNoe, 2020-09-11:
I would like to reopen this for further discussion, as the currently used algorithm turned out to be forbiddingly slow.

@GernotMaier @moralejo Could you comment on which algorithms are used in EventDisplay and Mars for this?

I have several ideas what kind of algorithms could be used here and how one could implement them in a fast way.

I think this should be a configurable component to allow for comparisons between different algorithms.

GernotMaier, 2020-09-11:
Eventdisplay uses the 'rob' option from TGraph::Fit():

https://root.cern.ch/doc/master/classTGraph.html#aa978c8ee0162e661eae795f6f3a35589

MaxNoe, 2020-09-11:
Thanks, this seems to be this method: https://en.wikipedia.org/wiki/Least_trimmed_squares

moralejo, 2020-09-13:
MARS uses also ROOT's "robust" fit option. We use "ROB=0.85", which means that at most 15% of the points (if found to be outliers) are discarded.  
Note: by default ROOT requires a minimum of (npoints+nparams+1)/2  of the original points to be used in the final fit.

MaxNoe, 2020-09-13:
Okay, this indeed sounds like the least trimmed squares algorithm, I'll look into it.

MaxNoe, 2020-09-15:
> which means that at most 15% of the points (if found to be outliers) are discarded.

Minor nitpick, but if I understood the algorithm correctly, it always uses the 85 % points with the lowest squared error, it performs no real outlier detection.

adonath, 2020-09-24:
@maxnoe and @Karl Kosack For outlier robust fitting I have successfully used the RANSAC algorithm in the past as well. It is one of the standard algorithms for feature point matching in computer vision. There is an implementation available in scikit-learn: https://scikit-learn.org/stable/auto_examples/linear_model/plot_ransac.html.  Maybe it's worth a try...

MaxNoe, 2020-10-02:
For now we go with the same approach as the ROOT base frameworks, which should also improve comparability of the DL1 stage. 

Implemented in #1419.

I am closing this for now again. But a more comparative study should be done including e.g. RANSAC and checking both computation performance and physics performance.

PR #1351 (2020-05-21): use robust fit to get timing slope
fixes #1275 

This just re-computes the slope with `scipy.stats.siegelslopes()` to get a result that is less affected by outliers (which is known to improve the TwoPassWindowSum image extraction method)

It still uses polyfit to estimate errors, since siegelslopes does not return a cov matrix.
closed 2020-10-02 by MaxNoe