cta-observatory/ctapipe #1277: Sub-array definitions

GernotMaier, 2020-04-14:
For the analysis of sub arrays from simulations, we usually pick a sub-set of the existing telescopes for the analysis.

How are those sub-arrays defined? 

I am asking because:
1. a large number of subarrays have been defined for different studies with Eventdisplay and Mars. This information should be accessible for ctapipe (and other systems)
2. telescope IDs used in the simulations change with productions (e.g., is MST5 not the same telescope in prod3, prod2, or prod4-MST)
3. reconstruction should start using the same telescope IDs as colleagues on the ground (i.e., we should already now make sure that a discussion about MST5 is the same telescope in simulations as on the architects drawings).

I am currently make sure to have consistency with the engineers and propose for now (as this will be replaced by a data base) the following two schemes:

1. a list of telescopes with their names, positions, and IDs in the different simulation sets (see attached example). This file needs to be updated after each new simulation

2. a list of (sub arrays), which allows then to simply map the telescope numbers in the subarrray definitions for the different simulation production.

Would that work? The above scheme would allow to 'hide' all the messy details about telescope IDs in the simulations from ctapipe. 

(and somehow github doesn't allow to attach yaml or ecsv files, this is why I've added a .txt)
[telescope_positions_south.ecsv.txt](https://github.com/cta-observatory/ctapipe/files/4473645/telescope_positions_south.ecsv.txt)
[layouts_south.yaml.txt](https://github.com/cta-observatory/ctapipe/files/4473647/layouts_south.yaml.txt)

kosack, 2020-04-14:
Following #738 , we will at some point move to having a set of configuration data per "prod", and then we can just include these subarray definitions as a table that one loads.  Since we have so many different types of simulation files right now, it's a bit tough -even in Prod3B, I seems to have files with very different telescope subsets and locations (at least as test files - perhaps the bulk of the data is not like that?)

kosack, 2020-04-14:
The best would be to have a way to query a simulation file and know what prod number or variant it is, and then we can map things easily.  Is that stored anywhere?  If not, I worry this is a fragile system and would break easily if somebody uses a file where the tel-ids are not as expected.

GernotMaier, 2020-04-14:
I don't think there is an explicit identifier for the production in the simulation files, although it is for all grid produced files encoded in the file names (e.g., /gamma_20deg_0deg_run103___cta-prod3-lapalma3-2147m-LaPalma.simtel.gz).

The idea here is that as soon as you know the prod number,  you are able to retrieve any pre-defined sub arrays (the simulations files usually are for at least the baseline configuration) without having the figure out again how the telescope identifiers are. You are also able to retrieve data for a certain telescope (e.g. MST5).

Again, I assume that this is a temporary solution to make the subarrays available currently considered. At some point we will only simulate those telescopes which will actually be built and we introduce the same telescope ids/names in the whole CTA world (to my knowledge is system engineering still discussing how components should be called).

MaxNoe, 2020-04-14:
Would it be possible to encode this in the simtel runheader in the future?
Some kind of production identifier.

There are two string fields in the run header that could be used for this, target and observer.

These are in the moment set to 
```
 'target': b'Monte Carlo beach',
 'observer': b'pcta15@jux83.zeuthen.desy.de'
```

Cerntainly target could contain something more useful than "Monte Carlo Beach". Although that sounds like a nice place to be :rofl:

GernotMaier, 2020-04-14:
What is 'it' in your question?

- we certainly should encode the production number into the header
- subarrays cannot be encoded, as this is a 'subset' of telescopes chosen after the simulation is done

MaxNoe, 2020-04-14:
I meant something identifying production. Like the string in the filename or something similar.

GernotMaier, 2020-04-14:
Yes, this should be done (but this point is mostly independent of my questions on subarray definitions).

MaxNoe, 2020-04-14:
So there are currently no subarray definitions inside ctapipe or ctapipe extra I think.

The way to select a subarray is to set the allowed_tels for the event sources and I think people manually looked up which tel ids they wanted to analyze for this.

I know Kai needed to reprocess a fair amount of data because he selected a non-sensical subarray (8 LSTs or something)

kosack, 2020-04-15:
> So there are currently no subarray definitions inside ctapipe or ctapipe extra I think.
> The way to select a subarray is to set the allowed_tels for the event sources and I think people manually looked up which tel ids they wanted to analyze for this.

yes, so far in Protopipe, there is a lookup table for mapping the subarrays, it's not done in ctapipe.  But that is also a bit of a hack and will break when we change prods (or try to analyze some other variant of the current prods), so having a prod identifier other than "Monte Carlo beach" would be great and simplify life, at least for future prods.

orelgueta, 2020-12-08:
> Would it be possible to encode this in the simtel runheader in the future?
> Some kind of production identifier.
>
> There are two string fields in the run header that could be used for this, target and [...]

Gernot pointed me to this point in the issue, so I thought I would comment.
There are a few ways to tell which production the file comes from (I am sure Konrad has more more), but none of them is "pretty":

1. In Prod5, Konrad added the release to this field
`'target': b"Monte Carlo beach, version='2020-06-28 15:07:23 CEST (konrad@wizard4)', release='2020-06-28 (candidate for full prod5 baseline setup)', eventio_version='2020-05-12 17:01:23 CEST (konrad@wizard4)'"`
You could extract the production info from there and if empty assume it's prod3b (or prod4). Very ugly and breaks easily, but provides some information.

2. Konrad added a parameter "ARRAY_CONFIG_NAME". It's not included in the info read out as far as I can tell (i.e., hessioxxx/include/io_hess.h), but it's in the header of each output simulation file, so I assume it can be added also to the i/o. Not very useful now, but could be in the future.

3. The safest bet is probably the parameter "detector_prog_vers" in hessio/pyeventio. This returns the sim_telarray version (times a 1000?). The version number changes also when the configuration changes, so the value should be exclusive and exact for each production. You could, for now, hardcode a dictionary that translates between the version and the production. For example,

- Prod3b - detector_prog_vers = 1481293431

- Prod5 - detector_prog_vers = 1593349643

kosack, 2020-12-09:
SO to summarize here: 
1. yes, please add a configurable prod name to the data format (that can be chosen by the person configuring simtelarray)
2. for now, we can make a "guess" function like we do with camera types (using the detector_prog_vers" to start with)
3. yes, also please provide some structured file (JSON, YAML) that has the mapping of subarray_name to tel_ids_in_subarray for each PROD.    We can put that on the data server just like other test files, and then we can add a functions like  `get_tel_ids_for_subarray(prod="PROD2", subarray_name="H1BFQ2") → List[int]`, and `get_subarray_names_for_prod(prod="PROD2") → List[Str]`

kosack, 2020-12-09:
Another related question: there are some runs from e.g. Prod3b that have a "super array" in them, and some where there are only a subset of telescopes (La Palma Reference Sims, for example).  Those would both identify as "Prod3b", but I guess the telescope-IDs are not the same, so subarray mappings would not work.  Is that correct? We could consider using *detector_prog_version* + *num_tels* as part of the `guess_prod()` function, I guess. Is there a list somewhere?

closed 2022-07-27 by kosack