cta-observatory/ctapipe #1278: Do not use astropy sky coordinates for ground and tilted ground frame

MaxNoe, 2020-04-14:
These are not sky coordinates in any sense, so they shouldn't be using astropy frames and `SkyCoords`

maxnoe, 2020-10-26:
I think we should use 3d coordinates in a well defined earth frame for this, e.g. WGS84 / ITRS, e.g. like astropy does for obsgeoloc, see

kosack, 2023-02-23:
I think we can close this now that we have the ability to transform GroundFrame to EarthLocation  #2167

maxnoe, 2023-12-03:
One of the unintended, surprising side effects was reported by @mdpunch  in #2478

mdpunch, 2023-12-05:
Hi,

I don't know much about the esthetics of the different containers (though I'd have thought that Cartesian relative to the centre of the array would be simplest, with then the centre of the array being defined in a fancier way... whatever).

But if keeping with SkyCoords, it would be nice to give an error if the user does inapplicable things, like using `separation` with the frame being `GroundFrame` (which gives an angle in reponse)?

I have my answer, I'd just like others not to fall in the trap...

// Michael.

maxnoe, 2023-12-05:
> The main issue is that having a nice container like SkyCoord that aggregates frames and attributes of those frames is nice (even if we currently misuse or fail to use it in many places).

I think we get more weird behavior by using something that has the explicit assumption of describing a fixed point in space to describe something fixed on Earth's surface, as can be seen by the astropy issue linked above.

We can re-use the representations and EarthLocation to get something better, with the behavior we need an none of the weird inherited behavior from skycoord without much effort I think.

mdpunch, 2025-03-12:
Found this (and the my linked issue) again, while googling to search for how to get the actual time delta between ground and tilted frames.  I had forgotten the use of `.cartesian`, but using it does indeed work:
```
pointing = AltAz(az=az, alt=alt)
subarray_tilted = subarray.tel_coords.transform_to(TiltedGroundFrame(pointing_direction=pointing))
subarray_tilted_distance = subarray_tilted.cartesian-sub_4LSTs_4MSTs.tel_coords.cartesian
subarray_tilted_norm = subarray_tilted_distance.norm()
from astropy.constants import c
sub_4LSTs_4MSTs_tilted_norm/c
```
Maybe add some information on the `subarray.cartesian` in the docs, e.g. here: https://ctapipe.readthedocs.io/en/stable/auto_examples/tutorials/coordinates_example.html and/or in https://ctapipe.readthedocs.io/en/stable/api/ctapipe.coordinates.GroundFrame.html#ctapipe.coordinates.GroundFrame and https://ctapipe.readthedocs.io/en/stable/api/ctapipe.coordinates.TiltedGroundFrame.html ?

maxnoe, 2025-03-12:
@mdpunch Sounds good. You could also think about directly contributing such algorithms in ctapipe, i.e. the low-level functions and a reconstructor only relying on the times and array locations.

mdpunch, 2025-03-12:
I suppose I could!  Maybe as part of a coding sprint (to help if I get stuck deciding where to put things)? Are there coding sprints for ctapipe?

mdpunch, 2025-03-12:
> Found this (and the my linked issue) again, while googling to search for how to get the actual time delta between ground and tilted frames. I had forgotten the use of `.cartesian`, but using it does [...]

(Never mind the calculation of distance, just using the z in the tilted frame is sufficient... I fall into this trap every 8 years or so!)  See https://github.com/cta-observatory/ctapipe/issues/2712#issuecomment-2718703117 for better implementaion.